UnchartedBull/OctoDash #2589: Dependance Issue - Armbian 21.08.6 Bullseye (Debian) - Odroid xu4 (32bit)

jalanjarosz, 2022-01-18:
**What doesn't work?**
Install script fails with: 
```
Package libappindicator3-1 is not available, but is referred to by another package.
This may mean that the package is missing, has been obsoleted, or
is only available from another source

E: Package 'libappindicator3-1' has no installation candidate
```

**What did you already try?**
downloaded install script and manually installed all dependencies.
Received same error when manually installing via:
`sudo apt-get install libappindicator3-1`

Added * to installation and received the following output:
```
pi@odroid:~$ sudo apt-get install libappindicator3-1*
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
Note, selecting 'libappindicator3-1' for glob 'libappindicator3-1*'
0 upgraded, 0 newly installed, 0 to remove and 0 not upgraded.

```
Apparently Debian may have depreciated this library in Deb10.   My OS is based on Deb11

**General Information:**
 - Hardware: Odroid xu4 (8 CPU big.LITTLE @ 2Ghz with 2gb RAM and 32GB eMMC)  Waveshare 7" HDMI LCD (H)
 - OS Info: 32bit Armbian 20.08.6 Bullseye (Debian 11) Kernel 5.4.160 (freshly installed)
 - OctoDash Version Latest via wget install script
 - OctoPrint Version 1.7.2

**Additional context**
Add any other context or screenshots about the feature request here.

jalanjarosz, 2022-01-18:
Attempted process as outlined in: https://github.com/UnchartedBull/OctoDash/issues/759

Received the following:
```
pi@odroid:~$ wget -O octodash.deb $releaseURL -q --show-progress
octodash.deb        100%[===================>]  58.18M  25.0MB/s    in 2.3s
pi@odroid:~$ sudo dpkg -i octodash.deb
Selecting previously unselected package octodash.
(Reading database ... 63595 files and directories currently installed.)
Preparing to unpack octodash.deb ...
Unpacking octodash (2.2.0) ...
dpkg: dependency problems prevent configuration of octodash:
 octodash depends on libappindicator3-1; however:
  Package libappindicator3-1 is not installed.

dpkg: error processing package octodash (--install):
 dependency problems - leaving unconfigured
Processing triggers for hicolor-icon-theme (0.17-2) ...
Processing triggers for desktop-file-utils (0.26-1) ...
Processing triggers for mailcap (3.69) ...
Errors were encountered while processing:
 octodash
pi@odroid:~$
```

jneilliii, 2022-01-18:
Appears you can install the old version possibly with a manual install script.

https://www.reddit.com/r/debian/comments/pn1oia/what_happened_to_libappindicator31_in_debian_11/

jalanjarosz, 2022-01-20:
I ran across the same post as you did.

Upon looking further into that post, I had to download 2 BUSTER packages:
`https://packages.debian.org/buster/libindicator3-7`
followed by:
`https://packages.debian.org/buster/libappindicator3-1`

Then install them in order noted above using:
`sudo dpkg -i <package name>`

Finally, the wget installation of OctoDash was able to be successfully executed.

At this point, I'm getting some sort of segmentation fault when X (ratpoison) is starting.  Unfortunately, the console is not providing any useful info, and there is no data in /var/log/Xorg.0.log (it's a 0kb file).

I'll have to dig further, but if anyone has suggestions, I'd appreciate it as well.

jneilliii, 2022-01-20:
At one point @kantlivelong was working on a service file only implementation removing ratpoison completely. Doing something like that might help in your case.

jalanjarosz, 2022-01-23:
I decided to downgrade the OS to: Armbian_21.08.6_Odroidxu4_buster_current_5.4.160.

OctoDash does install successfully on this build with the exception of "touch" is not functioning on my 7" Waveshare display, which is a different issue all together. X does recognize the touch device, it looks like a driver is loading as well.

ATM, I'm not sure if @UnchartedBull would like to keep this open to address Debian11/Bullseye issue as a future enhancement or not.  It's your call, otherwise I'm okay to close the issue.

closed 2022-04-19 by stale[bot]