UnchartedBull/OctoDash #2604: How to debug screenSleepCommand and screenWakepCommand?

meermanr, 2022-01-21:
**What doesn't work?**

OctoDash never put the display to sleep, even though the command I've given to `screenSleepCommand` works perfectly when called from a shell.

**What did you already try?**

I'm using a Raspberry Pi 4 with one of its native (Raspberry Pi branded) touch screens. The screen + xserver combo doesn't support DPMS, so I wrote a little python script to turn the screen on and off by manipulating the values the backlight brightness value at `/sys/devices/platform/rpi_backlight/backlight/rpi_backlight/brightness`.

The script is executable by all users, and works just fine when I call it:

```bash
/home/pi/rpi_backlight off
/home/pi/rpi_backlight on
```

So I changed my `~/.config/octodash/config.json` to end as follows:

```json
...
                        "pollingInterval": 2000,
                        "touchscreen": true,
                        "turnScreenOffWhileSleeping": true,
                        "turnOnPrinterWhenExitingSleep": false,
                        "preferPreviewWhilePrinting": true,
                        "previewProgressCircle": false,
                        "screenSleepCommand": "/home/pi/rpi_backlight off",
                        "screenWakeupCommand": "/home/pi/rpi_backlight on"
                }
        }
}
```

I was hoping there'd be an error logged somewhere, and poked around the files in `~/.config/octodash` hoping to find a general logfile. I then tried running a bare xserver (+ratpoison) and launching octodash myself from a terminal so I can watch its output, but all I see is:

```
pi@church:~ $ DISPLAY=:0 octodash
libGL error: failed to create dri screen
libGL error: failed to load driver: vc4
libGL error: failed to create dri screen
libGL error: failed to load driver: vc4
libGL error: failed to create dri screen
libGL error: failed to load driver: vc4
libGL error: failed to create dri screen
libGL error: failed to load driver: vc4
```

How can I check it is really calling this command? And as which user? With which current directories? Is it using exec(), execve(), system(), or passing it to `/bin/sh -c`?

**General Information:**
 - Raspberry Pi 4 + Raspberry Pi branded 7" touchscreen
 - OS Info: Raspbian GNU/Linux 10 (buster)
 - OctoDash Version: v2.1.2 (because the latest version only works on a fresh install: restarting it causes it to get stuck in some endless loop waiting for OctoPrint.. unless I delete `~/.config/octodash` and _then_ restart)
 - OctoPrint Version: v1.7.2

**Additional context**
OctoPrint and OctoDash are running on different machines on the LAN if that matters.

I installed OctoDash using the excellent little installer.

jneilliii, 2022-01-22:
wonder if you can update the command to include a pipe to a log file, ie `/home/pi/rpi_backlight off >> /home/pi/backlight.log`?

meermanr, 2022-01-29:
Good idea, but it doesn't appear to produce any files (not even empty ones). I start to suspect it is not actually calling the commands _at all_ which is why I was asking how to debug this.

I was hoping for a log from OctoDash showing the command it is trying to execute, and perhaps any output generated by it..

meermanr, 2022-01-29:
I also tried using the following but no files are created:

```json
pi@church:~ $ tail .config/octodash/config.json
                        "touchscreen": true,
                        "turnScreenOffWhileSleeping": true,
                        "turnOnPrinterWhenExitingSleep": false,
                        "preferPreviewWhilePrinting": true,
                        "previewProgressCircle": false,
                        "screenSleepCommand": "/usr/bin/date >> /dev/shm/sleep_cmd.txt",
                        "screenWakeupCommand": "/usr/bin/date >> /dev/shm/wakeup_cmd.txt"
                }
        }
}
```

UnchartedBull, 2022-02-22:
Unfortunately there are no log files written. I just tested this on a local dev build and it seems like everything is working just fine:
[image]

OctoDash uses `childe_process.exec` to execute these commands. To debug them I'd recommend:
- create a new js file with the following contents:
```
const exec = require('child_process').exec;

const tmp = exec(<your command>);
tmp.stdout.on('data', function (data) {
  console.log(data);
});

tmp.stderr.on('data', function (data) {
  console.log(data);
});
```
- execute that file which should give you a log output
- once the commands are working correctly they should also work on OctoDash.

taintedkernel, 2022-03-13:
I have not gotten a chance to debug it yet, but hitting the same issue after upgrading to v2.3.1 recently.  The only other change I did was enable spoolmanager.

taintedkernel, 2022-03-20:
I'm not too familiar with node, but assuming I did it correctly I saw no output in the console - the screen did turn off though:

```
pi@octoprint:~ $ cat debug_octodash_sleep.js 
const exec = require('child_process').exec;

const tmp = exec('DISPLAY=:0 xset dpms force standby');
tmp.stdout.on('data', function (data) {
  console.log(data);
});

tmp.stderr.on('data', function (data) {
  console.log(data);
});
pi@octoprint:~ $ nodejs --version
v10.24.0
pi@octoprint:~ $ nodejs debug_octodash_sleep.js 
pi@octoprint:~ $
```

taintedkernel, 2022-03-20:
And now as I go back to check on this later, it seems the screen sleeping is working.  I didn't make any changes to the rPi other then create the test JS file and install node.  Seems like it's an intermittent problem.

@meermanr do you have any updates for the issue on your end?

taintedkernel, 2022-03-21:
Checking back today, the issue is back again.  Thinking about it more, I believe the problem may be caused by the sleep condition never being triggered in the first place, rather then the sleep command not being executed correctly.  On my side when the screen doesn't sleep, I don't see "Ssshhh OctoDash is sleeping", but instead see the normal OctoDash UI with the status of "operational".

I'll try looking through the code some more to see if I can figure out why this might be the case, assuming my theory is correct.

jneilliii, 2022-03-21:
Octodash only goes to sleep if the printer is disconnected, unless you configure a custom action button for sleep and have my companion plugin installed.

taintedkernel, 2022-03-22:
Ahhh...  that makes a lot more sense now!  I might have missed that in the docs.  Either way, I do have the companion plugin installed and will look at making an custom sleep action button.  Thank you!

closed 2022-04-19 by stale[bot]