UnchartedBull/OctoDash #2608: OctoDash keeps reconnecting if API key is incorrect

queengooborg, 2022-01-25:
After replacing my old OctoPrint username with a new account (and revoking all the prior API keys), I noticed that OctoDash kept saying that it was attempting to reconnect to OctoPrint.  It would be great to instead get an on-screen message indicating that the API key is invalid, with instructions on how to correct it (or a one-tap fix).

UnchartedBull, 2022-02-22:
This would require quite a lot of work and I don't think that this feature will be used by many users. Won't work on this, sorry.

queengooborg, 2022-02-22:
Forgive me for my ignorance as I am not as familiar with Octodash's inner workings, but is it really that difficult to implement?  I mean, shouldn't Octodash return some form of error indicating an authorization failure, like an HTTP 401?

UnchartedBull, 2022-02-22:
This was mostly related to the one-tap fix. I've added a notification if 403 is returned which should make debugging a lot easier.

PR #2693 (2022-02-22): Show unauthorized error
Fixes #2608.
closed 2022-02-22 by UnchartedBull