UnchartedBull/OctoDash #2610: freezes and crashes on PI zero 2, something changed.

phenomeus, 2022-01-26:
**What doesn't work?**
dash 2.2.0 started to lag and crash on "PI Zero 2" this started around 1 month ago, I had to uninstall octodash,
when I go into the file menu, it needs a lot of time to show any files. when I print something, it suddenly freezes the whole system and the print is crap. 
I know that it worked flawlessly, somehow I have a feeling that maybe there was an update / code change and suddenly it started to behave like that. 

**What did you already try?**
i uninstalled octodash, kept everything up2date and printed without errors. reinstalled octodash again and its happening again. 

**General Information:**
 - Hardware PI Zero 2
 - OctoPi 0.18.0
 - OctoDash Version 2.2.0
 - OctoPrint Version 1.7.3 Python 3.7.3

UnchartedBull, 2022-02-22:
There was a minor memory leak in this version (which doesn't affect anything with 1GB RAM or more). The next release (hopefully released in the next couple of hours) will fix this, please report back whether that fixes your issue.

phenomeus, 2022-03-24:
sorry for late response, I reinstalled newest version (around 3 weeks ago) and tried a print. 

it happened again but lucky as I am at 99% somewhere, so I could still use the print. it just froze and the printer stuck at a position. temperatures for bed and nozzle were still at high printer level (dangerous). 

so its still not 100% back to normal. is there anything I can help with to debug it?

phenomeus, 2022-03-24:
I checked and I was on 2.3.0, now I wanted to update to 2.3.1 from octodash settings menu on the screen and got that error. 
i will upgrade through ssh now 
[image]

phenomeus, 2022-03-31:
my ram usage on an ender 3v2 with 2.3.1 octodash and 1.8.0 octopi on a zero 2 

pretty high doing nothing other than home and preheat. even Swap is mostly full. 

any chance do reduce it or is the combination of octodash and octoprint on a zero 2 too much? 

[image]

phenomeus, 2022-03-31:
@UnchartedBull during 5h print with 2.3.1, until now its solid, ram and swap is full, cpu cores are bored.
[image]

UnchartedBull, 2022-04-07:
RAM usage is pretty high due to electron mostly. That'll eat a good 60MB or something to launch Chrome, Octoprint also takes quite a nice chunk plus around 100MB or so for other random stuff that might be running. So having 500MB of RAM is really stretching it here. But OctoDashs RAM usage should not increase (unless there is a bug). So let's hope that everything will work from here on. Please reopen if this issue occurs again.

closed 2022-04-07 by UnchartedBull