UnchartedBull/OctoDash #2649: OctoDash doesn't start Octolapse

Mech131, 2022-02-07:
**What doesn't work?**
OctoDash doesn't start Octolapse (when the plugin is enabled) when i start a print from the screen.

**What did you already try?**
Ran the general update script for OctoPi, OctoPrint, and Octolapse

**General Information:**
 - Raspberry Pi 400 Rev 1.0
 - OctoPi 0.18.0
 - OctoDash Version [Not sure how to check this one]
 - OctoPrint Version [v1.7.3]

Mech131, 2022-02-07:
OctoDash Version v2.2.0

jneilliii, 2022-02-08:
Pretty sure starting a print from OctoDash should spawn the same actions when started from OctoPrint as OctoDash just tells OctoPrint to start the print. The only time that doesn't happen I think is if you are printing from SD card.

Mech131, 2022-02-08:
I'm not printing from the SD card. The OctoLapse screen pops up, but it doesn't start octolapse. For now, i've turned off Octolapse.

Thinkersbluff, 2022-02-15:
Do you have Octolapse configured to preview its plan before starting the print? I see that prompt in Octoprint but not on Octolapse, on my system.

UnchartedBull, 2022-02-22:
This seems more like a configuration issue. Please make sure that Octolapse isn't showing any prompts before the print as @Thinkersbluff mentioned. If the issue persists please reopen.

closed 2022-02-22 by UnchartedBull