UnchartedBull/OctoDash #2672: "Repeat print?" or "start print" stay on screen

bolsoncerrado, 2022-02-14:
**Is your feature request related to a problem? Please describe.**
I run a farm so sometimes I just want to repeat the last print I completed. Octodash by default shows the FILE/FILAMENT/CONTROL screen instead. 

Also sometimes I run Octoprint from the LAN and preload a file, "ready to press print", then when I go to Octodash I miss how to print that file that is already preselected using octoprint on the LAN.

**Describe the solution you'd like**
Perhaps these solutions are already aboard or controlled via an external plugin?

**Link other projects**
TouchUI ex-user. This was somehow "allowed" by TouchUI, which basically was a compact web interface with the same options as Octoprint.

jneilliii, 2022-02-14:
Octoprint supports the start action command which just prints the currently selected file. 

https://docs.octoprint.org/en/master/features/action_commands.html

bolsoncerrado, 2022-02-15:
I can just prest start over the computer on the LAN but wanted it integrated in the OctoDash UI...

UnchartedBull, 2022-03-02:
Yes this is definitely a nice feature. Have been using KlipperScreen until OctoDash is natively ported, that is the single best feature from TouchUI / KlipperScreen that isn't present in OctoDash. Will come in a future update :)

bolsoncerrado, 2022-06-17:
I just find out either the MK3S does not support action commands or if it does, you must have RPi ON, which I dont want as I'm using a Pi3 connected to the USB port...

So ANY clues on how to start a print job from an Octodash button?
Which command or action or what does happen when you press PRINT on the Octoprint interface? 
I just need to replicate that!

Thanks @jneilliii @UnchartedBull