UnchartedBull/OctoDash #2707: Control Panel "triangle" too small to be touched by finger

SaKiEQ, 2022-02-24:
**Describe the bug**
The disclosure triangle for the control panel is too small, or too close to the display border - on a 5" touch screen ' to be operated 
by my average finger. With a stylus is can be made to work... 

**Screenshots**
If applicable, add screenshots to help explain your problem.

**General Information:**
 - Hardware [Raspi 4]
 - OctoDash Version [2]
 - OctoPrint Version [latest]

UnchartedBull, 2022-02-24:
By disclosure triangle you mean the one that should close the notification center?

You can open and close the notification center by swiping down to open (start at the top) and up to close (start from the button). This should be the preferred way, the triangle icons are meant as a fallback.

SaKiEQ, 2022-02-26:
Hi, i tried swipe - seems to not work on my display (5" Waveshare". My display or configuration seems to like only touch, no 'gestures'. Have to dig in to see of this a a configuration issue. 

is the Notification center (that I call control center since it hosts the control buttons), supposed to work now also during printing  ?

UnchartedBull, 2022-03-02:
Yes Notification Center should be openable (is that even a word?) from every screen, including standby and printing.

closed 2022-03-02 by UnchartedBull

SaKiEQ, 2022-03-03:
I am pretty certain I could not open it while printing - even when using stylus...

PR #2724 (2022-03-02): Increase notification center icon hitbox
Fixes #2707