UnchartedBull/OctoDash #2714: 403 forbidden

mtbrown4ayl, 2022-02-27:
**Describe the bug**
I repeatably get a get error of 403 forbidden and it appears to be related to an API key.

**To Reproduce**
Steps to reproduce the behavior:
This is random and not consistent.

**Expected behavior**
This should not happen

**Screenshots**
If applicable, add screenshots to help explain your problem.

**General Information:**
 - Hardware [Raspberry Pi]
 - OS Info [I do use OctoPi]
 - OctoDash Version [v2.3.0]
 - OctoPrint Version [ v1.7.3]

**Additional context**
I REALLY like Octodash! Thank you! I have been using Octodash for about 4 weeks now and I have seen a couple of updates come through for Octodash but none of them seem to fix my problem. I have 2 different machines both running similar setups. I keep getting a "403 forbidden" error dues to what looks like an API key issue. I have replaced the API key 3 times now in both printers and continue to get and 403 forbidden screen on the locally attached touch screen. Octoprint still works via a local computer -Octoprint web screen. And the way I fix it is to "restart Octodash" on the pulldown tab on Octoprint. I'm not sure why this is happening. I'm not sure if it is something I'm doing, overlooking, or a plugin add-on causing the issue. Please help. Thank you!

UnchartedBull, 2022-03-02:
Do both machines have the same hostname by any chance? Please post your configs (remove the API Key beforehand). Thanks!

mtbrown4ayl, 2022-03-02:
Maybe. Are you asking about the "hostname" found in the raspi-config feature? Then yes they are both set as "octopi".

Thanks Mike

UnchartedBull, 2022-03-02:
And which URL did you enter in OctoDash? if they both are using `octopi.local` it might try to access the other instance instead of the one it's supposed to.

mtbrown4ayl, 2022-03-02:
Yes. I used octopi.local. So can I just change let's say one the to octopi1 and the other octopi2? And this change should fix the problem? Thanks Mike

UnchartedBull, 2022-03-02:
Yes definitely. Or use `localhost` if OctoDash is running on the same Pi that OctoPrint is running on

mtbrown4ayl, 2022-03-03:
So, I understand you correctly I would use octopi.localhost instead of octopi.local? Correct?
Thanks Mike

jneilliii, 2022-03-03:
no, literally use `localhost` or `127.0.0.1` in the place of `octopi.local`.

mtbrown4ayl, 2022-03-03:
Got it! Thanks for your help! Mike