UnchartedBull/OctoDash #2717: OctoDash hangs & fails to boot 80-90% of the time

FLYN-001, 2022-02-28:
Hi there, on startup OctoDash hangs the entire system & OctoPrint fails to finish loading & I can't connect to the web browser while the OctoDash initialisation is taking a long time screen is up & the system is hanging. It can sit like this for 30 minutes or more if left.

I have to Ctl Q OctoDash & sudo service OctoPrint restart to get OctoPrint running & then I have to turn of additional plugins like OctoEverywhere & Octopod & then restart the Pi.

I seems like OctoDash is starting up & trying to connect to OctoPrint before its fully loaded & running due to these additional plugins extending Octoprint's launch time as they must connect online servers.

Without these plugins running Octoprint & OctoDash both start perfectly well each time.

Is there a way to make OctoDash only startup AFTER Octoprint is fully loaded & up & running? 

As this issue is really starting to get me down, its taking me 20-30 minutes each day to just get me printer running & then when I do I can't use the additional online services I need. I really want to use OctoDash but its seriously limiting the functionality of my printer right now & eating a whole lot of my time trying figure out why its doing this.
Please can you help with this, I'd be most grateful.

Thanks.

Pi4 2g
latest software, plugins & octopi image

UnchartedBull, 2022-03-02:
That seems really weird. I was also running a Pi4 with Klipper, Octoprint and Marlin and had no issues whatsoever. I see a few possible fixes:

- start with a fresh image (best on a new SDCard and check whether the issue still occurs)
- Check your `~/.xinitrc` to check what is loaded during startup
  - this is also where you'd put a sleep command so it looks something like this (this should wait 30s to start OctoDash):
  ```
#!/bin/sh

xset s off
xset s noblank
xset -dpms

sleep 30

ratpoison&
octodash
  ```
- run htop or something if the Pi is in this weird state to check what process is causing this issue.

FLYN-001, 2022-03-02:
Hi! thanks so much for the reply. 
I have tried fresh images multiple times & sadly not once has it helped. 😖
Funny you should mention that. I put a 30 second sleep in that file today. It too sadly didn’t help. 

The pi does seem to be doing something weird as when it hangs it won’t appear on my WiFi network until after I ctrl Q octodash & sudo service OctoPrint restart when it eventually works & I can login on the web browser

FLYN-001, 2022-03-02:
I have installed another fresh image as you released 2.3.1 this evening. So I’ve tried that. 

I also increased the physical separation between the pi & the DSI screen. A combination of those two things seems to have cured the issue. 

I will use the unit for a few days & see how it goes. Fingers crossed it’s sorted now. 

Love the new icons btw. 😉