UnchartedBull/OctoDash #2720: Printer information impairs touch responsiveness

hzzm028, 2022-03-02:
A few days ago the GUI on my Pi Touchscreen offered me an Octodash update. I concurred.
Since the update I see every printer notification as an animated box, hiding the interesting information on my screen.
It's a permanent thing, like every other progress percentage pops up, heating progress etc. etc.
It never stops.

But that's not the main issue. Since this elegant box animation, the touchscreen responsiveness is down the drain.
Each action now takes around 3-5 seconds to react; be it simply navigation between menus or cancelling prints.

I've read through everything that ~/.config/octodash/config.json now offers, but I can't find anything that would let me disable the atrocious information box spam.

(Raspi 4B 8GB)

Swap-File, 2022-03-02:
Does a reboot temporarily fix your UI lag?

I've been noticing OctoDash UI lag that builds up over time.  It seems related to the number of printer information popups I get. 

If I run a print with 800+ layers, I'll get 800+ printer notification popups in OctoDash and then the OctoDash UI will start to crawl.   I get this many notifications because I have my slicer set to generate a notification for each layer (so my printer attached LCD can display it).

If I just leave Octodash running with no prints going, the UI seems to stay responsive.  Also, if I reboot, the UI lag goes away.

hzzm028, 2022-03-02:
> Does a reboot temporarily fix your UI lag?
Yes, today after a reboot it acted totally normal.
At the moment (after 3 hours of printing) it starts to feel somewhat slower.
Will report back this night after ~10-12 hours of printing.

UnchartedBull, 2022-03-02:
Oh that is interesting, but can be possible. at the moment notifications aren't cleared (I didn't saw a reason since there were only a few usually). The next update will remove the info notifications from OctoDash again since they don't seem to be too useful.

I'm going to do some tests locally and see if I can identify the core issue.

closed 2022-03-02 by UnchartedBull

Swap-File, 2022-03-02:
I don't mind the popup notifications, but it would be nice to adjust how long they stay on the screen and be able to set them up to self clear.

I do really like how this latest update enabled being able to display the camera while printing!  Thanks for getting that working.

jneilliii, 2022-03-02:
Was that added as a core feature or is that part of my plugin? If it's in core features now I could remove that from the plugin.

hzzm028, 2022-03-04:
I see you (@UnchartedBull) trying to push an update, but on a factory octopi (Linux octopi 5.10.63-v7l+ #1496 SMP Wed Dec 1 15:58:56 GMT 2021 armv7l GNU/Linux) it doesn't run. Error in the attached picture.
Sorry I sounded a bit grumpy there to begin with.

The truth is, I take your tool for granted in my everyday recreational workflow, forgetting that there is a noble volunteer behind it, putting in all that effort to maintain it. Thank you.
[image]
My quick assumption according to https://github.com/UnchartedBull/OctoDash/issues/2727 is, that i need to run the install script again.

jneilliii, 2022-03-04:
Running the manual update script should work and fix this bug at the same time.

PR #2722 (2022-03-02): fix annoying notifications
printer info notification are now just shown in the bottom right and won't trigger a notification anymore

should fix #2720