UnchartedBull/OctoDash #2727: Javascript Error when trying to update to 2.3.1

kats-greg, 2022-03-03:
I have the red dot on the gear indicating an update is available, but when I click on the update to 2.3.1 button, I get a javascript error message saying something about got is not a function.

BigJoy75, 2022-03-03:
I also got the same error message

kennethnilssen, 2022-03-03:
> I also got the same error message Finally I managed to update using script

Could you please share how?

fredlex3dprints, 2022-03-03:
I get same error trying to update from 2.3.0 to 2.3.1

LaPe63, 2022-03-03:
Same error here trying to update from 2.3.0 to 2.3.1 from the GUI (app.asar Type error: got is not a function)
I had to run the install script to update: 
bash <(wget -qO- https://github.com/UnchartedBull/OctoDash/raw/main/scripts/install.sh)

BigJoy75, 2022-03-03:
> > I also got the same error message Finally I managed to update using script
> 
> Could you please share how?

You can see in how to do it in [https://github.com/UnchartedBull/OctoDash/wiki/Update](url)
 

wget -qO- https://github.com/UnchartedBull/OctoDash/raw/main/scripts/update.sh | bash

spiff72, 2022-03-03:
same issue here - but using the script update after SSH into the pi worked fine.  Last update I did via the UI worked fine, but this one failed with the javascript error.

SaKiEQ, 2022-03-03:
same here.. SSH'ing my way into it right now :)

SaKiEQ, 2022-03-03:
the wget method worked flawless on one machine with a Pi4, on a Pi1A+ it fried the X Server....  I guess it was about time to setup this Pi from scratch anyway. 

[Xorg.0.log](https://github.com/UnchartedBull/OctoDash/files/8178596/Xorg.0.log)

RichardvanGestel, 2022-03-04:
the 'wget' command also worked fine here on a pi3:

pi@octopi3:~ $ wget -qO- https://github.com/UnchartedBull/OctoDash/raw/main/scripts/update.sh | bash
Updating OctoDash
octodash.deb        100%[===================>]  59.89M  4.73MB/s    in 13s
[sudo] password for pi:
(Reading database ... 93823 files and directories currently installed.)
Preparing to unpack octodash.deb ...
Unpacking octodash (2.3.1) over (2.3.0) ...
Setting up octodash (2.3.1) ...
Processing triggers for hicolor-icon-theme (0.17-2) ...
Processing triggers for desktop-file-utils (0.23-4) ...
Processing triggers for mime-support (3.62) ...
Done. Restart your Raspberry Pi to start the newest version!

morphias2004, 2022-03-07:
Same Java error.

bolsoncerrado, 2022-03-07:
Same error here. A screencap:
[image]

SW0rden, 2022-03-10:
It would seem that using the script no longer works.  Tried a couple of times and nothing is updated and get the same JS error as the above post.

jneilliii, 2022-03-10:
what happens when you run the command 

```
wget -qO- https://github.com/UnchartedBull/OctoDash/raw/main/scripts/update.sh | bash
```

SW0rden, 2022-03-10:
Thanks, got it!  I had left off the "| bash". Without it, it seemed as if it was working and asked about rebooting, but just fell back to the prompt and nothing was updated.

bolsoncerrado, 2022-03-10:
> wget -qO- https://github.com/UnchartedBull/OctoDash/raw/main/scripts/update.sh | bash

On a working environment with 2.3.0 if I ssh the above, I get:

Updating OctoDash
wget: missing URL
Usage: wget [OPTION]... [URL]...

Try `wget --help' for more options.
dpkg: error: cannot access archive 'octodash.deb': No such file or directory
rm: cannot remove 'octodash.deb': No such file or directory
Done. Restart your Raspberry Pi to start the newest version!

Any c ues?

bolsoncerrado, 2022-03-10:
OIC [](https://github.com/UnchartedBull/OctoDash/wiki/Troubleshooting#fix)You've most likely exceeded the GitHub Rate Limit, therefore the newest release can't be retrieved from the server.

Used manual method.

closed 2022-04-19 by stale[bot]