UnchartedBull/OctoDash #2728: Scrolling long names?

bolsoncerrado, 2022-03-03:
**Is your feature request related to a problem? Please describe.**
To find a way to display long filenames by scrolling or...? 

**Describe the solution you'd like**
Scroll long filenames or find a way to show the full filename of a gcode file. 

**Additional context**
Like Prusa Mini screen does for example?

sleake, 2022-12-11:
I second this. Or maybe just have it wrap in someway so you can see the full name. I use long names to contain some info that makes a difference if there are multiples:

`<object name> -<printer> -<location on the bed> -<other key details> -<print duration>`

So they would look something like:
`PI_TFT 5_0 Case -CR10 -C -60s -filchg3 -3h42m.gcode`
`Big vase v4 -E3v2 -FL -195c -50s -vase -5h2m.gcode`

Do you have any plans on when 3.0 will be released? Is there a list of what things you plan on adding in 3.0? (Great work on this BTW. Very useful/helpful!)

bolsoncerrado, 2022-12-11:
@sleake I ended up using the suggested theme here: https://github.com/UnchartedBull/OctoDash/issues/3178

closed 2022-12-11 by bolsoncerrado