UnchartedBull/OctoDash #2739: web cam image too small on Octodash screen

squiddy1234, 2022-03-06:
Hi everyone, firstly i am very new to Octodash and 3d printing in general. My problem is that i have managed to get Octodash to display my logitech webcam image on my ELECROW 7'' screen by adding a 'custom action' and choosing the [!WEB]http://octopi.local:80/plugin/octodashcompanion/webcam    option from the command dropdown list, however the address displayed when copying and pasting directly from the dropdown command box is [!WEB]http://octopi.local/?#   . the image is very good but it occupies only about a sixth of the screen. could anyone tell me what to add to the command line to get the webcam image full screen? Is there a full screen button somewhere?

jneilliii, 2022-03-06:
The webcam option is part of my plugin not Core OctoDash so you may want to close this issues and open one in the plugin's repo instead. As for size, if you are using the address `[!WEB]http://localhost:5000/plugin/octodashcompanion/webcam` the camera should be filling the screen 100% because of the css being used in the page that is being loaded [here](https://github.com/jneilliii/OctoPrint-OctoDashCompanion/blob/master/octoprint_octodashcompanion/templates/webcam.jinja2).

squiddy1234, 2022-03-07:
thanks for the advice, by the way still no luck with the webcam even with your suggested address

closed 2022-03-07 by squiddy1234

TheGreatKahedin, 2024-12-15:
> thanks for the advice, by the way still no luck with the webcam even with your suggested address

Hey, i'm having the same issue, did you manage to solve it?