UnchartedBull/OctoDash #2745: Wake from sleep and turn tplink kasa on

jasharrar, 2022-03-09:
Hi
Firstly love octodash. Looks great and so far working brilliantly. 
I bought a tplink kasa 105 (for anyone reading this the newer tapo ones don't work, but the kasa ones are in Argos for £8.99 right now).  I have managed to get the power off button to work in the UI with the PSU control settings. I have the system set to sleep and I can wake it with a tap in the display. But I can't work out how to get the PSU (kasa) to switch on when I do this so I can start the system. Any clues?
[image]
Many thanks

jasharrar, 2022-03-09:
[image]

jneilliii, 2022-03-09:
Not sure why those images aren't coming through, but I can tell you that OctoDash will only use on of plugin options for waking from sleep, so if you have PSU Control I believe it will take priority over any of the other power plugin options.

jasharrar, 2022-03-09:
Thanks, also not sure but it's just the "octodash is sleeping" image do nothing interesting.  I will try removing PSU control and see if that helps thanks

jasharrar, 2022-03-10:
Okay so I disabled PSU control plugin. But it still doesn't power up the printer when I tap the "shhh! Octodash is sleeping" screen.
Any other ideas?

jneilliii, 2022-03-10:
And you configured the setting in octodash for tplinksmartplug plugin?

jasharrar, 2022-03-10:
[image]
yep i think so

jasharrar, 2022-03-11:
jneilliii you are a legend.  It was the config file, opened, went to bottom, changed entry to true. full reboot and i worked first time :)  thank you so much
[image]