UnchartedBull/OctoDash #2790: Dead socket and unresponsive pi when accessing files menu

reelpilot, 2022-03-24:
**Describe the bug**
I've had this issue now on multiple pi's running octoprint / octodash and the issue appears to be intermittent. Essentially when attempting to access the files page, sometimes the screen gets stuck trying to load the page. After a few minutes I can go back to the main page, but I cannot utilize any function and 'socket is dead' appears on the screen. The touchscreen then becomes inoperable on subsequent reboots.

**To Reproduce**
When starting a print I first go to the control page and preheat the printer. Then I go to the files page to navigate to the file I want to print, however intermittently the file page never loads even if I walk away for hours. At this point I cannot access OctoPrint whatsoever via the IP address. On the touchscreen I can press back to return to the main page where 'socket is dead' is now displayed in the lower right. If I navigate back to the control page and attempt to shutdown it can take anywhere from 5 minutes to several hours for the pi to shutdown. However upon reboot the touchscreen, a HyperPixel 4" 800x480, is completely non responsive to any commands, but otherwise displays normally. At this point I can access OctoPrint via the IP address and otherwise run my printer normally. Thus far I have been unsuccessful in restoring the touchscreen function and my only fix has been to reimage the pi. 

**Expected behavior**
When accessing the files page it should load the files and not crash the system.

**Screenshots**
[image]
shots**

**General Information:**
 Raspberry Pi 4 4GB
 - OS Raspbian Buster 
 - OctoPi  Version 0.18.0
 - OctoDash  Version 2.3.1
 - OctoPrint  Version v1.7.3

**Additional context**
Add any other context about the problem here.

UnchartedBull, 2022-04-07:
Do you have a lot of files on your OctoPrint instance by any chance? It seems like OctoPrint is dying trying to send OctoDash information regarding the files. Could you maybe temporarily move some files of the OctoPrint server and check whether that issue still persists.

reelpilot, 2022-04-07:
I'm beginning to think that it does have to do with the files as you suggested. I do have lots of files, in excess of 450, that are organized in various folders since I have lots of products that I sell. I recently noticed a pattern of the same 4 instances having this issue, but the majority of the instances I have not experienced this. Upon closer examination I discovered that a few files had not synced over to those 4 instances, but they were fully synced on the others. So my guess is that somehow that was triggering the lockup. A few days ago I verified that all files are synced across all instances and thus far I have had no issues.

electroklas, 2022-11-03:
Octodash

Have similar issues, and some of this sounds familiar.

* New octoprint install, vanilla from raspian imager.
* Had pi on dhcp, and did the install of octodash and plugins.
* Changed net, and set a fixed ip, (no more inet access)
* Had issues with finding the octoprint instance from the octodash setup via its dns name… changed to use 127.0.0.1 instead… (prob defective DNS setup in my local net)
* Connected to octoprint webgui from pc and uploaded gcode file, and gui becomes somewhat unresponsive (both web, and i beleive also octodash), no sign of file list update.
* Ordered restart of system from webgui
* File list now showed uploaded file
* Started print, all is looking good
* X hours later about to check status by accessing webgui from new host on the lan
* The webgui is loaded  to some degree..  but very slowly… but filelist and other data is not updated..
* Looking at octodash display it updates temperatures and layers, but states: socket is dead.
* Print is still going on, unaffected

(Print will hopefully continue for another 23h , (47 in total)

sharanchius, 2024-01-03:
I also have the same problem. What about fix?

electroklas, 2024-01-03:
My problems were probobly due to missing dns functionality, and no internet access... Had my octopi unit on a separate network segment that used a win10 pc as network connection via dual eth cards and routing... Reconfigured the network layout and installed a fw/router insted of pc. This led to full DNS functionality, and all is working as supposed..