UnchartedBull/OctoDash #2792: Can't uninstall (or disable via Companion Octoprint plugin) the OctoDash plugins

bolsoncerrado, 2022-03-25:
**Describe the bug**
I'm facing some screen issues with some of the plugins I added when I installed Octodash so I want to either disable or remove such plugins

**Expected behavior**
Plugins to be disabled or removed.
For example, I installed Octodash companion but no matter if I uncheck the plugins and reboot, the plugins are still there.

**Screenshots**

[image]
[image]

**General Information:**
 - RasPi 3B+ on OctoPi image 0.18
 - OctoDash Version 2.3.1
 - OctoPrint Version 1.7.8

jneilliii, 2022-03-25:
This is not the companion plugin repo, I manage that separately in its own repo. However, I'd you disable the plugin in plugin manager and restart octoprint it should go away. In very rare instances you may need to reboot the pi completely or force refresh your browser using shift+f5 if any plugin isn't doing right.

bolsoncerrado, 2022-03-25:
@jneilliii you mean the octoprint plugins menu installs an outdated version of the companion? how do I install yours?

Thanks!

jneilliii, 2022-03-25:
Pretty sure the installation script installs the latest version of my plugin.

bolsoncerrado, 2022-03-25:
OK then I missunderstood you. What did you mean by "This is not the companion plugin repo, I manage that separately in its own repo. "?

bolsoncerrado, 2022-03-25:
And this is as ridiculous as it gets. I went ahead and completelly uninstalled Octodash.

Then I reinstalled it with ONLY Companion+Preheat plugins.

Not only the same Companion version shows up that it's also DIFFERENT from the one I posted above this morning AND the Layer plugin is still there and not gone!

WTH?

[image]

[image]

bolsoncerrado, 2022-03-25:
Besides the CANT UNINSTALL PLUGINS issue, I think the companion from the install script and the companion from the octoprint plugins list ARE different.

That or @jneilliii updated it between this morning's post and this one above :?

jneilliii, 2022-03-25:
I have not made any updates today for the plugin.

bolsoncerrado, 2022-03-25:
Awesome then.

jneilliii, 2022-03-25:
Are you just trying to get rid of the plugins? That's done from the octoprint plugin manager settings.

bolsoncerrado, 2022-03-25:
No way, really!? LOL I thought octodash plugins were exclusively handled via octodash companion/octodash command line

jneilliii, 2022-03-25:
No, those are all in octoprint. The install script is just doing command line installs in octoprint as a convenience.

bolsoncerrado, 2022-03-25:
No clue then why I got two different plugin views on the companion then.

Thanks tho.

closed 2022-03-25 by bolsoncerrado