UnchartedBull/OctoDash #2799: Cannot invoke Enclosure output that triggers a shell script

TomFrost, 2022-03-30:
**What doesn't work?**
The Enclosure plugin supports shell scripts as outputs. I use this to control the LED strips that light my enclosure. After following the syntax to invoke one of these custom outputs from Octodash, nothing happens when tapping the icon.

**What did you already try?**
Enabled the Enclosure plugin in the OctoDash config. The custom action is:

```json
{
  "color": "#6be0db",
  "command": "[!OUTPUT]4,on",
  "confirm": false,
  "exit": false,
  "icon": "lightbulb"
}
```

This is the output (with the ID of 4) that I'm trying to invoke. Note that this works in the native Enclosure plugin UI:
[image]

**General Information:**
 - Hardware [e.g. Raspberry Pi, if you have layout issues please also include your screen resolution]: **RPi 3B**
 - OS Info [e.g. Raspbian Buster Lite, please also indicate if you used OctoPi]: **OctoPi**
 - OctoDash Version [e.g. v1.0.0]: **2.3.1**
 - OctoPrint Version [e.g. v1.0.0]: **1.7.3**

**Additional context**
Add any other context or screenshots about the feature request here.

FWIW, I'm aware that the Action Commands plugin is the recommended way to execute shell scripts from OctoDash. That's inconvenient, however, because I may want to activate my lights when the printer isn't currently accepting GCode input, such as when I'm taking a new mesh.

Thanks for the lovely utility!

jneilliii, 2022-03-30:
You might want to check out GCODE System Commands plugin. Those get run as soon as the gcode command gets sent from OctoPrint and doesn't have the same delay you are seeing with Action Commands.

TomFrost, 2022-03-31:
No joy I'm afraid -- I ran another `G29 T` and fired off my new gcode commands to control the lights, and had the same effect as the Action Commands plugin: The commands queued up for execution after the printer became idle.

jneilliii, 2022-03-31:
Well that might depend on the command it comes after. G29 is one of those blocking commands, so for your use case having a generic system command custom action might make more sense.

closed 2022-04-19 by stale[bot]