UnchartedBull/OctoDash #2800: Reconfigure API key and Hostname through Terminal

Forsyth-Creations, 2022-03-31:
## Issue
Hi all! I just recently started using OctoDash, and I'm already loving it over the TouchUI. However, I've hit a snag. Long story short, I went through the on-screen setup, but I connected it to the wrong OctoPrint instance (I have several on-network). Now, after rebooting, it is having issues initializing. I receive a 403 error, while waiting on the OctoDash start screen. I'd like to change the API key to reflect the proper machine (It would be weird if my Taz GUI started controlling my Ender).

## Debugging
I've tried restarting the Pi, but I know the API key stored in OctoDash is for the wrong printer, so I don't know if it is having issues because of that. 

## Here is my setup
 - Raspberry Pi 4
 - OctoPi used for OS installation 
 - OctoDash Version 2.3.1
 - Using a touchscreen as my interface. Still trying to get touchpoints to rotate with xrandr, but that's not an OctoDash issue
 
I look forward to hearing from you all! I hope it's just as simple as ssh-ing into the pi and changing a settings file. If not, I'm ready for some debugging.

Forsyth-Creations, 2022-03-31:
Slight update to my issue, I realized your docs mention a config.json file at ~/.config/octodash/. I have changed the hostname and API key there, but I'm still having initialization issues. The initialization screen is complaining that CORS isn't enabled, but I have checked to make sure it is.

jneilliii, 2022-03-31:
You restarted both octodash and octoprint after making those changes?

Forsyth-Creations, 2022-03-31:
Sorry, I meant to update this! Changing the host name to http://localhost/ fixed the problem. Initially I put in my IP (since it's set to static on my network), but that didn't work. I put it in as 'http://192.###.#.###". We're all good to go now, though! I'll close the issue out. Any chance you have any tips on Pi4 screen rotation? Seems they're using new drivers, so the usual display_rotate way isn't working for me. I'm thinking about disabling xrandr and going with the old ways. I know this isn't an OctoDash issue, just thought I'd ask!

jneilliii, 2022-03-31:
You can do this with xorg configs typically. This might be helpful: https://community.octoprint.org/t/inverting-an-hdmi-display-on-the-raspberry-pi-4b/22804?u=jneilliii.

Forsyth-Creations, 2022-04-01:
I was able to get it fully working by commenting out the xrandr driver, and just going old school. I hope that helps people in the future! That means I used the following command

```
sudo nano /boot/config.txt
```

Then commented/added the following lines as seen below and it started working:
```
#dtoverlay=vc4-fkms-v3d
display_rotate=3
```

I believe you can also just use the following instead of ```display_rotate```
```
lcd_rotate=2
```

While using display_rotate, you will also have to apply a rotation matrix to your screen (or at least that's what I did). That would be the following commands:
```
 cd /usr/share/X11/xorg.conf.d
sudo nano 40-libinput.conf
```

Then edit this section to add the rotation matrix:
```
Section "InputClass"
        Identifier "libinput touchscreen catchall"
        MatchIsTouchscreen "on"
        Option "TransformationMatrix" "0 -1 1 1 0 0 0 0 1"
        MatchDevicePath "/dev/input/event*"
        Driver "libinput"
EndSection
```

Anyway, that was very convoluted/probably could've been done simpler, but wanted to have this here for anyone who needs it. Anyway, OctoDash is great. My roommate already wants to buy a touchscreen and a pi just to do it to his printers. And he's been giving me a hard time about OctoPrint for a year now! Great work!

closed 2022-04-01 by Forsyth-Creations