UnchartedBull/OctoDash #2847: Plugin and main screen

Nico4589, 2022-04-14:
Hi, it's the first time I use the raspberry and this interface for my 3d printer.  I have no idea how I can install the plugins?  also I would like to set the second extruder and the chamber temperature on the main screen.  How can I do?  
I'm going crazy.

Thanks for your help

metasean, 2022-04-16:
I'm also a new user, so I can't fully answer your questions, but —if you're also using OctoPrint— you should check out Octodash Companion.

1. In OctoPrint, click the wrench icon in the top-rightish area.
2. On the left hand side, under OCTOPRINT, click "Plugin Manager".
3. At the top of the plugin manager pop-up, "+ Get More".
4. Search for, then install, "OctoDash Companion".

Now, from the left hand side of your OctoPrint page, under "PLUGINS", you can select the "OctoDash Companion".  From there, you can manage your plugins, custom actions, and a few other things.

Nico4589, 2022-04-18:
Thanks for your help!!  But can I also change the image of the icons?

jneilliii, 2022-04-18:
My companion plugin will let you change the icons of custom action buttons. To change any other icons on the main screen would require using a custom css theme more than likely, and when then not sure if it's possible.

Nico4589, 2022-04-19:
How can i set the language in italian?

Absolucy, 2022-11-11:
Insert activity here, because stale bots suck.

closed 2025-03-18 by queengooborg