UnchartedBull/OctoDash #2887: Bed level visualizer

gerthalberg, 2022-04-30:
Enabling the octoprint Bed level visualizer would be a really nice addition to the capabilities

jneilliii, 2022-05-01:
You can create a custom action button to load a heatmap version of the BLV plugin and update the mesh. Use the below as the command to use. 
```
[!WEB]http://127.0.0.1:5000/plugin/bedlevelvisualizer/bedlevelvisualizer
```

gerthalberg, 2022-05-02:
Thank's for the hint - really appreciate the response.
Entered as above I don't get a result, but when I use a browser - I do get a nice map when I use port 80 i.e. don't specify a port instead of port 5000. Is that correct or is there something amiss in the Pi/octoprint/octodash installation?

jneilliii, 2022-05-02:
If it works without the 5000 port specified then use that for sure. The port 5000 will only work with localhost/127.0.0.1 for the address in a default octopi image implementation.

atax112, 2023-10-04:
Hi. 
I have set the custom command in my octodash plugin settings, when using the address from my PC the heatmap does show, but I cannot update the mesh, however from octodash on the Pi screen, using the command does nothing.

It asks if im sure I want to execute the command but nothing happens when I confirm.

What am I doing wrong? Is it possible to trigger BLV from the octodash screen? 

Thanks.

jneilliii, 2023-10-04:
Does your settings look like this?

[image]

atax112, 2023-10-04:
yes, pretty much
[image]

i also tried with my octoprint IP and no port but still nothing

atax112, 2023-10-05:
Ok, so when I execute the command from the notification/action screen it actually loads the heatmap, but the update mesh button does nothing, same as from PC. 

From the Controls menu, when I execute the command it only says loading webpage in the notification screen and nothing happens

jneilliii, 2023-10-06:
> From the Controls menu, when I execute the command it only says loading webpage in the notification screen and nothing happens

Yeah, this is a known bug I've seen posted about any web page embed. Only seems to work from the notification panel for some reason. 

> but the update mesh button does nothing

enable debug logging in the BLV plugin's settings, restart OctoPrint, try the process from the notification panel and then check plugin_bedlevelvisualiver_debug.log and octoprint.log for errors/hints at what might be going on.

atax112, 2023-10-07:
> enable debug logging in the BLV plugin's settings, restart OctoPrint, try the process from the notification panel and then check plugin_bedlevelvisualiver_debug.log and octoprint.log for errors/hints at what might be going on.

from octoprint.log:

`2023-10-07 16:15:49,415 - octoprint.plugins.octodashcompanion - INFO - loaded data from /home/pi/.config/octodash/config.json: {'config': {'octoprint': {'url': 'http://ataxprint.local:80/', 'accessToken': '84BB78B1B2FF42B5AA45420AC649E71A'}, 'printer': {'name': 'N3', 'xySpeed': 150, 'zSpeed': 5, 'disableExtruderGCode': 'M18 E', 'zBabystepGCode': 'M290 Z', 'defaultTemperatureFanSpeed': {'hotend': 190, 'heatbed': 6, 'fan': 100}}, 'filament': {'thickness': 1.75, 'density': 1.25, 'feedLength': 450, 'feedSpeed': 120, 'feedSpeedSlow': 3, 'purgeDistance': 30, 'useM600': False}, 'plugins': {'displayLayerProgress': {'enabled': True}, 'enclosure': {'enabled': False, 'ambientSensorID': None, 'filament1SensorID': None, 'filament2SensorID': None}, 'filamentManager': {'enabled': True}, 'spoolManager': {'enabled': False}, 'preheatButton': {'enabled': True}, 'printTimeGenius': {'enabled': True}, 'psuControl': {'enabled': False, 'turnOnPSUWhenExitingSleep': False}, 'ophom': {'enabled': False, 'turnOnPSUWhenExitingSleep': False}, 'tpLinkSmartPlug': {'enabled': False, 'smartPlugIP': '127.0.0.1'}, 'tasmota': {'enabled': False, 'ip': '127.0.0.1', 'index': None}, 'tasmotaMqtt': {'enabled': False, 'topic': 'topic', 'relayNumber': None}}, 'octodash': {'customActions': [{'color': '#e1b12c', 'command': 'M140 S60', 'confirm': False, 'exit': True, 'icon': 'fire-alt'}, {'color': '#0097e6', 'command': 'M140 S0; M104 S0', 'confirm': False, 'exit': True, 'icon': 'snowflake'}, {'color': '#ff0000', 'command': 'M104 S195', 'confirm': False, 'exit': True, 'icon': 'fire'}, {'color': '#dcdde1', 'command': '[!WEB]http://10.0.0.182/plugin/bedlevelvisualizer/bedlevelvisualizer', 'confirm': True, 'exit': True, 'icon': 'ruler-horizontal'}, {'color': '#0e61b4', 'command': '[!WEB]http://127.0.0.1:5000/plugin/bedlevelvisualizer/bedlevelvisualizer', 'confirm': False, 'exit': False, 'icon': 'ruler-horizontal'}], 'fileSorting': {'attribute': 'date', 'order': 'dsc'}, 'invertAxisControl': {'x': False, 'y': False, 'z': False}, 'pollingInterval': 2000, 'touchscreen': True, 'turnScreenOffWhileSleeping': False, 'turnOnPrinterWhenExitingSleep': False, 'preferPreviewWhilePrinting': False, 'previewProgressCircle': True, 'screenSleepCommand': 'xset dpms force standby', 'screenWakeupCommand': 'xset s off && xset -dpms && xset s noblank', 'showExtruderControl': True, 'showNotificationCenterIcon': True}}}
2023-10-07 16:15:50,523 - octoprint.server.util.sockjs - INFO - User ataxprint logged in on the socket from client ::ffff:10.0.0.174
2023-10-07 16:15:52,831 - octoprint.server.util.flask - WARNING - lastmodified is not timezone aware, cannot check against If-Modified-Since. In the future this will become an error!
2023-10-07 16:15:53,100 - octoprint.server.util.flask - WARNING - lastmodified is not timezone aware, cannot check against If-Modified-Since. In the future this will become an error!
2023-10-07 16:16:14,362 - octoprint.util.jinja - WARNING - Loading plugin template 'bedlevelvisualizer_octodash.jinja2' from '/home/pi/oprint/lib/python3.9/site-packages/octoprint_bedlevelvisualizer/templates/bedlevelvisualizer_octodash.jinja2' without plugin prefix, this is deprecated and will soon no longer be supported.`

I tried several times after restart and to me it seems that the button doesnt even generate any logs?! 

BLV log doesnt change and since enabling debug before restart it only has this line:

`[2023-10-07 16:16:14,350] DEBUG: using stored mesh for octodash view: [['+0.027', '+0.030', '+0.005', '+0.105'], ['+0.072', '+0.010', '+0.025', '-0.013'], ['-0.025', '+0.012', '+0.070', '+0.040'], ['-0.053', '+0.062', '+0.115', '+0.137']]
`

atax112, 2023-10-08:
Digging around with my limited experience but I dont see anything out of the ordinary on the page, however the button/script wont trigger from either PC / Pi Screen

[image]

jneilliii, 2023-10-08:
I think I found the error. 

[image]

[image]

Will have to look and see how to resolve that on the BLV plugin, as CSRF validation was added in a recent release of OctoPrint,.

jneilliii, 2023-10-08:
Created the above issue on the BLV plugin repo and about to release v1.1.2rc7 to address the problem.