UnchartedBull/OctoDash #2944: Octopi switched to Ubuntu server for their image base. Octodash does not currently work on it.

supawiz6991, 2022-05-24:
I started talking about this a little in #2531 but it really deserves its own ticket.

Octopi has switched to Ubuntu server for their 64-bit builds. Currently there is not a stable version available, only nightly's [here](http://unofficialpi.org/Distros/OctoPi/nightly-arm64/). They have been developing this for a little over a year at this point.

Out of curiosity I ran though the install steps for Octodash on the latest nightly image and ran into the following:
1. gir1.2-gnomekeyring-1.0 no install candidate. This package must have been depricated and remove in ubuntu the same as it has in Raspberry pi os Bullseye. 
2. Ignoring the first issue, I finished the install steps and rebooted but only got an ubuntu desktop (the image is terminal only by default). See the Image I attached.
[image]

Octoprint and octodash can still be installed manually on Raspberry pi os Bullseye for 64-bit support for now but from what I gathered from the octopi issues, there is some stability problems.

supawiz6991, 2022-06-13:
Bump to keep this alive until until unchartedbull has a chance to respond.

alison-gravley, 2022-07-24:
In case this helps someone:

I can boot into Raspberry Pi OS 64-bit 2022-04-04 and my Hyperpixel 4.0 works perfectly with the new easier config options. Orientation is correct, touch is correct, etc. See [https://github.com/pimoroni/hyperpixel4/issues/177](url)
For completeness, I add this to the bottom of my /boot/config.txt to get the correct orientation I need. I do NOT uncomment the "dtoverlay=vc4-kms-v3d" farther up as I get a black screen otherwise.
 
`dtoverlay=vc4-kms-dpi-hyperpixel4`
`dtparam=rotate=270,touchscreen-swapped-x-y,touchscreen-inverted-x`

I manually install OctoPrint, go through initial configuration, and I am able to successfully auto-start it.

Getting OctoDash to even install hasn't entirely been straightforward, especially with the automatic install that I abandoned. I had to do some work to get the libappindicator3-1 installed by grabbing the .deb from debian.org as it is no longer included. When I did get the autoinstall "working", nothing I did would fix the screen orientation and touch input, assuming it is an issue with ratpoison. At this point, I started with a new image of Raspberry Pi OS 64-bit and started over.
See [http://ftp.us.debian.org/debian/pool/main/libi/libindicator/libindicator3-7_0.5.0-4_arm64.deb](url)
See [http://ftp.us.debian.org/debian/pool/main/liba/libappindicator/libappindicator3-1_0.4.92-4_arm64.deb](url)

Luckily, the manual installation method had little to no issues once I figured out what I had to do. I grabbed the v2.3.1 arm64.deb from github release and installed with dpkg.
`wget -0 octodash.deb https://github.com/UnchartedBull/OctoDash/releases/download/v2.3.1/octodash_2.3.1_arm64.deb`

I figured at this point that ratpoison was the problem and I needed to use mutter instead. I created the .xinitrc like the original instructions and made mine look like this:
```
#!/bin/sh

octodash &
exec mutter
```
Then in my bashrc I ended up using plain startx after the other way failed.

```
if [ -z "$SSH_CLIENT" ] || [ -z "$SSH_TTY" ]; then
    #xinit -- -nocursor
    startx
fi
```
At this point I rebooted and I have a functioning OctoDash. I have NOT put everything through its paces yet, but I am able to get through the configuration processes and the touchscreen works perfectly.

supawiz6991, 2022-08-30:
@alison-gravley So, we figured out how to get Octodash to work (including the automatic script) in this issue, -> https://github.com/UnchartedBull/OctoDash/issues/2531

This issue that you commented on isn't related to Raspberry pi os. The Octoprint team has opted to move to Ubuntu Server instead of continuing with Raspberry Pi OS and the point of this issue was to bring that to UnchartedBull's attention and see if they intend to add Ubuntu Server support or if its going to be Raspberry Pi OS only...though it I think a lot depends on Octoprint at this point on if they are going to stick with Raspberry Pi OS for their 32-bit builds or move completely to Ubuntu Server.

jneilliii, 2022-08-30:
> The Octoprint team has opted to move to Ubuntu Server instead of continuing with Raspberry Pi OS

I'm not sure this is true. Unless I'm mistaken the latest stable and RC versions of octopi are still using Raspbian/pi os.

supawiz6991, 2022-08-30:
> > The Octoprint team has opted to move to Ubuntu Server instead of continuing with Raspberry Pi OS
> 
> I'm not sure this is true. Unless I'm mistaken the latest stable and RC versions of octopi are still using Raspbian/pi os.

I just revisited the Octopi Github and there seems to have been a major change for RC2.

RC1 (See this [issue](https://github.com/guysoft/OctoPi/issues/770) ) - They switched the 64-bit build to Ubuntu Server.
RC2 (See this [issue](https://github.com/guysoft/OctoPi/issues/788) ) - They switched 64-bit build back to raspberry pi os though it still sounds like they are still on the fence as to which way they want to go and may yet still switch back to Ubuntu Server.

I don't envy being in UnchartedBulls position because the uncertainty of their direction is going to make it very difficult to know what to do. OctoDash does work in Bullseye but does need fixes to its auto install script and a verification of an apparent package name change (as discussed in this [issue](https://github.com/UnchartedBull/OctoDash/issues/2531) ).

Running this in ubuntu seems like it it will require some work to convert so I won't be the least bit surprised if this issue gets shelved for the time being until the Octopi Devs make a firm decision on which os they will go with.

Hillshum, 2025-03-21:
Current OctoPi uses Raspberry Pi OS Bookworm, and OctoDash 2.4.1 runs on it.

closed 2025-03-21 by Hillshum