UnchartedBull/OctoDash #2959: Use M125 as OctoDash Pause

FLYN-001, 2022-05-30:
Hopefully a quick easy one. 
would it be possible to have a checkbox to enable the OctoDash print pause button to make direct use of Marlin’s M125 park on pause feature?

…After it has been enabled in the printer’s firmware of course. 

As leaving the nozzle on the print for any length of time just goes to ruin the print. 

I’ve used some G code in OctoPrint’s G code scripts but it’s not as good as the native M125 feature where you get a chance to have fans disabled & to purge & prime a oozing paused nozzle just before resuming the print.

jneilliii, 2022-05-30:
isn't it just a matter of configuring Marlin's `ADVANCED_PAUSE_FEATURE` which is used by M600?

jneilliii, 2022-05-30:
Ah, so you don't have `host_action_commands` enabled, which would let Marlin tell OctoPrint that it is paused and resume once the process is completed?

FLYN-001, 2022-05-30:
> Ah, so you don't have `host_action_commands` enabled, which would let Marlin tell OctoPrint that it is paused and resume once the process is completed?

Yes host action commands is enabled. OctoDash pauses, OctoPrint sends the M125 command, the pops work on OctoDash for pause, purge more & continue. 

The printer pauses, parks, purges & resumes all through OctoDash - but you have to click the regular Octodash resume button on the UI to get back to the progress screen.

jneilliii, 2022-05-30:
Ah, I see what you mean now. OctoDash is not picking up the state change from paused to printing and automatically close that screen.

FLYN-001, 2022-05-30:
> Ah, I see what you mean now. OctoDash is not picking up the state change from paused to printing and automatically close that screen.

Yeah exactly that, even though it’s all working it trips up on that last step. So I just thought I’d there was a native M125 pause option it work a bit better. 😉