UnchartedBull/OctoDash #3065: OctoDash prevents OctoPrint from loading

jaymichaelic, 2022-07-16:
**Describe the bug**
OctoDash disabled the OctoPrint from booting on the same Raspberry Pi

**To Reproduce**
Steps to reproduce the behavior:
1. Flashed OctoPrint onto an SD Card using Raspberry Pi Imager
2. Booted and set up OctoPrint
3. Installed OctoDash using the install script
4. Enabled CORS and other defaults
5. Rebooted Raspberry Pi
6. OctoDash comes up on display and good with the Rsspberry Pi touch screen
7. Checked OctoPi web connection to find that there was no connection--no more OctoPrint!  SSH times out...no connection
8. Checked Router--still same IP and also same OctoPrint name comes up as the device on my network
9. OctoDash cannot connect to http://localhost/

**Expected behavior**
Since there is no mention in the documentation that TWO Raspberry Pi's are needed (one for OctoPrint and one for OctoDash), and no video mentions that that two are necessary, I presume that when the Raspberry Pi boots, OctoPrint boots in the background as it should since it is a service of the server, and OctoDash should be able to connect to the api through the localhost connection.

**General Information:**
 - Hardware Raspberrt Pi 4 Model B, Enviciv 7" Captaive Touch Screen for Raspberry Pi
 - OS Info Respberry Pi Imager v1.7.2 with OctoPi v0.18 
 - OctoDash Version v2.3.1 (installed using the github URL)
 - OctoPrint Version v1.8.1

**Additional context**
Intitally, because it wasn't very clear to me that the same Raspberry Pi can handle both OctoPrint and OctoDash, so I did purchase a second one.  However, before using it to the point where I cannot return it, I would like to figure out if I can indeed just use one Raspberry Pi.

jaymichaelic, 2022-07-16:
P.S.  I did this process twice before posting.

One thing I need to figure out is why I cannot SSH into the device when running OctoDash.  Is it simply because I haven't finished the setup because it cannot log into OctoPrint?

How can I stop OctoDash from launching at statup?  I am familiar with the Linux Kernel, but not an expert.

jneilliii, 2022-07-17:
I can at least confirm for you that running OctoDash on the same Pi as OctoPi/OctoPrint is flashed on works perfectly fine.

jneilliii, 2022-07-17:
the SSH access is enabled by default by OctoPi on initial boot, unless those options were changed in pi imager's advanced settings during the flashing process.

jneilliii, 2022-07-17:
if running on the same pi and you can access OctoPrint's web interface you can use the OctoDash Companion plugin I made to stop the service and return to the command line from the system menu.

jaymichaelic, 2022-07-17:
> the SSH access is enabled by default by OctoPi on initial boot, unless those options were changed in pi imager's advanced settings during the flashing process.

Right.  I am able to SSH after OctoPi is flashed onto the Raspberry Pi and after OctoPrint has loaded up as well.  No changes to the advanced settings disabled SSH.

The problem seems to be that once OctoDash is installed and it reboots, SSH is no longer available and neither is OctoPrint.

I will try installing the companion Plugin prior to installing OctoDash, but that still assumes that I would be able to get to OctoPrint after OctoDash is installed, which is the problem.

If OctoDash is indeed disabling SSH and OctoPrint somehow, as I suspect it is inadvertently doing, AND it automatically boots, it effectively holds my entire Raspberry Pi hostage because that's the only thing I see and/or can access.  I know that's not its intention, but it is the only explanation.

How can I disable OctoDash from starting automatically?  This may allow my Raspberry Pi to boot and initialize OctoPrint.  Then I can SSH or use a keyboard to start OctoDash manually from the command line interface.

UnchartedBull, 2022-07-18:
OctoDash will never touch your SSH configuration. That's completely off limits of any program tbh and I think requires privileged access, which OctoDash doesn't have. The only things OctoDash is doing during setup is setup autostart and the update script (if wanted by the user). You can find the relevant files here:

https://github.com/UnchartedBull/OctoDash/blob/main/scripts/install.sh#L790 (if you want to disable OctoDash from auto-starting you have to remove the `octodash` line form `.xinitrc`.

Can you still ping your OctoPrint raspberry once it's in that broken state?

And yes OctoDash is designed to being run on the same pi as OctoPrint. Multiple Pis are supported though.

jaymichaelic, 2022-07-22:
Just wanted to let you know I gave this all another go and it worked!  So happy to be using this brilliant piece of software.  Very grateful for the help.  I will contribute $ome funds as soon as I sort out my paypal account.

To give you a run down of what I did this time:

I decided to use the manual installation of OctoDash without automatically starting.  I must have missed this part the second time around.  I then confirmed that OctoPi was reachable through SSH and that OctoPrint could be accessed via the web, I enabled auto-login and finally the rest of the auto start instructions.

It still did not auto-detect the server, btw, but it got through the manual setting.

closed 2022-07-22 by jaymichaelic