UnchartedBull/OctoDash #3084: I have three screens and none of them will sleep

Mech131, 2022-07-27:
**Describe the bug**
I've tired multiple commands and cannot get my screens to sleep. I have EVICIV 7 Inch Portable USB Monitor Raspberry Pi Touch Screen IPS Display Computer Monitor 1024X600 16:9

**To Reproduce**
Steps to reproduce the behavior:
1. "turnScreenOffWhileSleeping": true
2. ScreenSleepCommand: "xset dpms force standby"
3. Screen stays on 24/7

**Expected behavior**
I expect the screens to turn off or go to sleep

**Screenshots**
None

**General Information:**
 - Hardware [Raspberry Pi 4B | EVICIV 7 Inch 1024X600 16:9]
 - OS Info [OctoPi / Octoprint]
 - OctoDash Version [I frequently update my device, I'm likely on the most recent version]
 - OctoPrint Version [v1.8.1]Hardware [e.g. Raspberry Pi, if you have layout issues please also include your screen resolution]

**Additional context**
[octoprint-systeminfo-20220727100015.zip](https://github.com/UnchartedBull/OctoDash/files/9200447/octoprint-systeminfo-20220727100015.zip)
[octoprint.log](https://github.com/UnchartedBull/OctoDash/files/9200449/octoprint.log)

jneilliii, 2022-07-27:
It's your printer going into a disconnected state are always stays connected? The sleep timeout requires octoprint to be in a disconnected state IIRC.

Mech131, 2022-07-27:
I thought the sleep timer was supposed to work while the printer was printing and when it is in standby mode. Even when my printer is turned off, the screen will stay on, and it just says to touch it to try and reconnect.

jneilliii, 2022-07-27:
Sounds like your screens might not support dpms properly for OctoDash to be able to control it. You might want to look into installing xscreensaver as a possible alternative.

Mech131, 2022-07-29:
@jneilliii XScreenSaver seems difficult to install. Do you know of any other alternatives?

jneilliii, 2022-07-29:
Really, I just did `sudo apt install xscreensaver` I think.

Mech131, 2022-08-01:
@jneilliii Small correction to what I said earlier. When I turn off the printer and OctoDash disconnects from the printer, the screen will shut off / go into screen saver mode. This would indicate that it working to a degree. How do I get it to sleep during a print so the screen goes off when i'm out of the room and I can just tap it to turn it back on and see the progress?

jneilliii, 2022-08-01:
you can't with the built in dpms control of OctoDash, it's what is managing the screen sleep I think.

Mech131, 2022-08-01:
Just to clarify and make sure I understand correctly. Using Octodash's screen saver, it will only activate when the printer is disconnected from Octodash?

jneilliii, 2022-08-01:
Correct

Mech131, 2022-08-02:
Screensaver is acting how it is supposed to. To get it to perform as I want, I needed to install XScreenSaver.

closed 2022-08-02 by Mech131