UnchartedBull/OctoDash #3087: Plugins support?

TheMiataDude, 2022-07-29:
**What doesn't work?**
Trying to use plugins like quick stop and plugins like exclude region, when I try to add them doesn't work. Was it designed this way? And if so im cool with that

 - OctoDash Version (I think 0.07 but that may have just been the octodash companion...)
 - OctoPrint Version (1.8.1)

jneilliii, 2022-07-29:
OctoDash only has limited support for specific plugins, it's not like touch UI where it's just rendering OctoPrint's interface differently.

TheMiataDude, 2022-07-29:
That’s about what I thought. Does anyone have a list by any chance that shows compatible plugins other than the octodash plugins?

jneilliii, 2022-07-29:
https://github.com/UnchartedBull/OctoDash/wiki/Octoprint-Plugins

closed 2022-07-29 by TheMiataDude