UnchartedBull/OctoDash #3129: No option for HTTPS octopi on the initial install startup of the Pi

tre4b, 2022-08-19:
**Describe the bug**
My Octoprint instance is setup using SSL and a local self signed certificate.   The Pi has the CA cert installed.  When running the script it fails to find the Octopi and the manual entry does not allow using an https url

**To Reproduce**
Steps to reproduce the behavior:
1) run the install scrip as detailed in the instructions.
2) reboot the pi
3) Follow onscreen prompts until you get to the point of manually entering a URL

once you get here with no ability to put the address in you are pretty much stuck.  There is no way to abort the process and going backwards eventually lead to a corrupt popup but did not resolve the issue.

**Expected behavior**
An ability to specify https://localhost within the installation startup process of the OctoDash program.

**Screenshots**
Not really possible.  I could not get out of the loop and did not think to photograph.

**General Information:**
Raspberry Pi 4 with yesterdays version of Octopi available through the Raspberry pi installer.  1.8.2 I believe but am currently rebuilding it to try again.

**Additional context**
I believed that putting Octopi on to ssl was best practice and saves the crossed out padlocks.  I had been using Octodash for quite some time up to that point.  I tried to change the URL in Dash Companion but could not get anything to work and in my attempts managed to corrupt the pi.  Am now rebuilding from scratch and would still like to use Octodash.  The install software looked much better than the challenges I had when I installed it last time.  It did however need me to have a keyboard to plug into the pi.   Perhaps the necessary input could be added to the script that I ran via the console.

jneilliii, 2022-08-19:
you could plug in a keyboard and manually edit the URL, or post install just modify the config.json file directly or use my OctoDash Companion plugin to update it from OctoPrint's settings.

tre4b, 2022-08-19:
The http part of the address seemed to be hard coded as it only let you add the stuff that comes after the setup.  Unfortunately the place the script runs blocked me from being able to access the device using a terminal to change an address manually.

My current plan is to temporarily disable HTTPS on the system and reinstall octodash, then I should be able to change the url in the companion app and reinstate the HTTPS stuff.

tre4b, 2022-08-19:
Well that got it working, swapping it back to https seems to be the challenge.  I have tried with the companion app but it did not work.  Now am trying to find config.json but not sure where to look for it....

jneilliii, 2022-08-19:
not sure why the companion app wouldn't work, but after making changes you have to restart octodash. the plugin adds a stop octodash and start octodash items to the system menu in OctoPrint by your username. the file can be found in /home/pi/.config/octodash/config.json.

tre4b, 2022-08-19:
I cannot get octodash to connect with octopi running on https.   I have changed the address in companion.  After the first boot it had changed back, so I changed it again.  Now it sticks but simply does not allow octodash to connect to octopi.

at a bit of a loss.

tre4b, 2022-08-19:
The app des seem to be making the changes, and I have just checked in config.json.  I did reboot after just to be sure and I have restarted a few times now but no luck.

the address it is using is https://localhost:443/   is that the right address?   I have tried the DNS address too and with and without the / on the end. (although that gets added back if I leave it off).

jneilliii, 2022-08-19:
so, I don't think you need the :443 at the end, but also since it's localhost you don't need to use https either because OctoPrint will just work if you use `http://localhost:5000`. The https stuff is being handled by haproxy, and that might not be listening on localhost addresses.

tre4b, 2022-08-19:
Yep, you are right.  So I made it difficult for myself because I assumed it had to be changed and I could have just left it at default!

All working normally on ssl with the above setting.

closed 2022-08-19 by tre4b