UnchartedBull/OctoDash #3164: Octo dash does not start, it stays starting.

Losky620, 2022-09-04:
After trying to make a raspberry camera work, due to a possible bad contact, Octodash does not start and therefore octoprint does not connect either.

I restarted several times, disconnected the camera and the printer and still it continues as in the photograph.

I have a raspberry 3+, latest version of octoprin and octodash.

[image]

Murphysboy, 2022-09-05:
Exactly the same issue here. happened today, no idea how to fix it. the API thing is already ticked, tried different cables and USB ports

ruralaccent, 2022-09-21:
I have hit this exact same issue. For me it started after updating to Octoprint 1.8.3. I have three identical octoprint installations and they all have this error after updating to 1.8.3.

queengooborg, 2025-03-18:
OctoDash v2.4.0 will have the settings menu added to the standby screen to allow for reconfiguration of OctoDash, which should help fix issues like this one in the future!

closed 2025-03-18 by queengooborg