UnchartedBull/OctoDash #3178: Reduce Font Size on File List (OR make names scrollable somehow)

bolsoncerrado, 2022-09-12:
**Is your feature request related to a problem? Please describe.**
We use large filenames in our 3D print farm, sometimes most of the characters are common except a few last words.

**Describe the solution you'd like**
Reduce Font Size on File List (OR make names scrollable somehow)
(user selectable?)

**Additional context**
Is there anyone able to help RIGHT NOW even if I have to edit system files until next version comes out and hopefully includes this solution??

Thanks!!

jneilliii, 2022-09-12:
This is my very hacky solution for getting filenames to scroll on hover, which is weird on touch screens... edit /home/pi/.config/octodash/custom-styles.css. It was originally based on the NOX theme, just can't remember if there were other changes I did to core OctoDash. There is a good chance of that...it's been a while since I did it. 

```
.files__object {
  background-color: #222222 !important;
  width: 92vw !important;
}

.files__object:hover .file__name {
  margin-left: -100% !important;
  display: inline-block !important;
}

.files__name {
  font-weight: normal !important;
  font-size: 3.1vw !important;
  transition: 5s;
  transition-timing-function: linear;
  /* start marquee changes */
  width: 100vw;
  overflow: ellipsis !important;
  position: relative !important;
  margin-left: 0 !important;
  display: inline-block !important;
  white-space: nowrap !important;
  transform-origin: right;
  transition: 5s;
  transition-timing-function: linear;
  /* end marquee changes */
}
```

bolsoncerrado, 2022-09-13:
Much appreciated.
If instead of scrolling I would just want to reduce font size, would that be possible?

Ty!

bolsoncerrado, 2022-09-13:
PLus it seems I have "issues" with the code provided hehe

https://www.dropbox.com/s/kza21a6oix53fj3/20220913_081206.jpg

jneilliii, 2022-09-13:
That's why I mentioned I started with the NOX theme. Your theme might not work the same, and there's a good chance that there were code changes to OctoDash to get it to work.

bolsoncerrado, 2022-10-05:
What about just reducing the font size of the list of files? Pretty pleeeeease??

@jneilliii  ?
@UnchartedBull  ?
Anyone ?

jneilliii, 2022-10-05:
Font size is in the css listed..

.files__name {
  font-size: 3.1vw !important;
}

But I don't know if it will help your situation again because the name clipping I think is happening server side.

bolsoncerrado, 2022-10-06:
Hey! It worked! Many thanks!

Will future octodash updates overwrite the customcss changes?

THanks

bolsoncerrado, 2022-10-06:
Also do you know the names of the size, weight and  time fields to update those fonts too?

[image]

jneilliii, 2022-10-06:
you probably could have just uploaded the [NOX theme](https://github.com/UnchartedBull/OctoDash/tree/main/themes/NOX) and gotten more real estate for the names too...

[image]

You'll see it hides the size and filament completely. Here are the relevant CSS for that.

https://github.com/UnchartedBull/OctoDash/blob/c1e03eefeb5a005772fd7688c456086e87916b75/themes/NOX/custom-styles.css#L166-L175

closed 2022-10-07 by bolsoncerrado