UnchartedBull/OctoDash #3183: Octodash cannot wake up the Tasmota and TPLink Poweradapter anymore since new install of 2 Octopis

TimRostock, 2022-09-14:
Hi,
here are 3 Octopis running with octodash and Tasmota and a TPlink Adapter, perfekt till the day before yesterday.
I had to new install 2 of them and after the install of octopi backup und install octodash all works fine. Except i cannot wake up the Tasmota and TPlink Adapter anymore. tried all, change raspis change network change dsi cable change sdcard. nothing is helping. the third octopi, who was not new installed runs perfekt till now, perfect wake up etc.
wake up with the octopi bolt in browser works perfect also for all 3!.

tried also new install from raspi pi imager and with etcher from iso.
anyone else?

jneilliii, 2022-09-14:
you have the tplink options enabled and configured inside octodash settings?

TimRostock, 2022-09-14:
yes. and for the other pi+octodash tasmota. works fine for over a year. works also in browser with the octo bolt. but no more with wake up after fresh installation

TimRostock, 2022-09-17:
And the standbye mode  for pitft50 and pitft70 is also no more working with the fresh installation. possible a driver problem between spi displays and octoprint?!