UnchartedBull/OctoDash #3197: Failure to initialise on OctoPi 0.18.0 & OctoPrint 1.8.3

ruralaccent, 2022-09-21:
I updated my three Raspberry Pi4 octopi installations today to Octoprint 1.8.3.

On the two installations where octodash wasn't connected to the printer, it failed to pass the initialisation screen when the system was restarted. 

On the installation where octdash was up and running and had been connected to the printer, it was displayed 'port is dead', and when I rebooted that installation, it then got stuck on the initialisation screen like the other two.

Octoprint functions as per normal and I can print just fine via octoprint, but octodash is now broken and won't ever finish initialising.

I all cases I used the prompt in octoprint to update to the latest version.

Al3601, 2022-09-22:
same issue after upgrading from 0.180 to 1.8.3

ruralaccent, 2022-09-22:
Following advice from Charlie over on the Octoprint forum, this issue relates to an API bug in 1.8.3 that prevents the global api working correctly. To resolve the issue (and future proof OctoDash as the global api is depreciated), follow these steps:

1. Set up an application api key for OctoDash in OctoPrint,
2. Update the OctoDash config file via SSH with that new application api key,
3. Reboot the system and OctoDash should initialise just fine.

cp2004, 2022-09-23:
The global API key usage should be fixed in the next bugfix release (1.8.4 I think), however now is a good time to realise that eventually the global API key will go away entirely and to switch to using an application key, or the API key specific to your user.

jneilliii, 2022-09-24:
Probably the same problem, you're using a global api key and not an application key in your OctoDash settings? You can edit that using the command `nano /home/pi/.config/octodash/config.json` via SSH.

lluisd, 2022-09-27:
> Probably the same problem, you're using a global api key and not an application key in your OctoDash settings? You can edit that using the command `nano /home/pi/.config/octodash/config.json` via SSH.

yes it was easy to solve it by doing that, but I didn't expect a breaking change by jumping to 1.8.3.

cp2004, 2022-09-27:
It wasn't a deliberate breaking change, and is now fixed with 1.8.4.

The global API key *will* be going away so it's a good reminder to not use it 😉

army11b1980, 2023-07-11:
I just did a fresh install and now I'm stuck on the welcome screen. It's like the touch is not responding anymore. I will try to do this solution.

queengooborg, 2025-03-18:
Touchscreen issue can be fixed via https://github.com/UnchartedBull/OctoDash/wiki/Troubleshooting#touchscreen-is-not-responding

closed 2025-03-18 by queengooborg