UnchartedBull/OctoDash #3198: Octodash with Rp4 & Hyperpixel4 goes green

sonypsx, 2022-09-22:
Hardware: RP4/4GB
Display:Hyperpixel 4.0
Powersupply: 4.0A+

Software:
Image: Clean Bullseye 64Bit build date 06. Sept 2022 incl apt-get update and upgrade
Manual setup of Octoprint 1.8.3
Octodash: 2.3.1 
While installing Ocotodash i run into to several issues e.g. rotation, missing libs like libappindicator3-1 but last but not least got that
all running
Solution of rotation found in: https://github.com/UnchartedBull/OctoDash/issues/2944
Solution for getting around of  libappindicator3-1 -> repack with libayatana-appindicator3-1 and "apt --fix-broken install"

Setting in /boot/config.txt for the display:
`dtoverlay=vc4-kms-v3d
max_framebuffers=2
dtoverlay=vc4-kms-dpi-hyperpixel4,rotate=270
dtparam=touchscreen-swapped-x-y,touchscreen-inverted-x`

What it looks like:
Boot of Bullseye is seen normal with correct colors, including rasperry logo
Octodashlogo in the correct color
Octodash gui for about 1 second in correct colors
and then bam everything is greenish and very hard to read (hulk smash) :)
Navigating, touch is working fine and responsive, but yeah in green!

Exactly the same issue was seen at: https://github.com/UnchartedBull/OctoDash/issues/1995

sonypsx, 2022-09-22:
Update:
After deploying, installing i think about 5 scratch installations i found out two important things:
1.) it is NOT octodash which causes the the display to go green
2.) it is octoprint 1.8.3 at least
As soon i start the octoprint service the screen goes greenish (also the desktop)

i will update this thread ...

sonypsx, 2022-09-22:
As promised Update:
I was able to figure out how it is 100% reproduceable.
If in octoprint 1.8.3 the plugin "octorelay v1.20" is enabled then this leads to the greenish screen
I hope this one helps others as well

jneilliii, 2022-09-23:
octorelay is a GPIO based plugin so it would make sense that it would cause issues with HyperPixel since that uses the GPIO to communicate with the screen. I suspect you have a setting in OctoRelay that is turning on one of the pins causing the display to do that.

sonypsx, 2022-09-23:
Yeah true, in fact i did another try with even no config at all behind octorelay - but it still harms HyperPixel4.
However - i deleted octorelay and i will never use it again :)

sonypsx, 2022-09-23:
Problem solved!

closed 2022-09-23 by sonypsx