UnchartedBull/OctoDash #3203: Touch not working on Official Raspberry Pi 7" display touchscreen

oobedoob, 2022-09-23:
I've got everything setup and Octoprint works fine as is, but when trying to setup OctoDash using my Raspberry Pi 3 with the 7"  Official Raspberry Pi display touchscreen when it boots into the OctoDash welcome screen I cannot do anything as it will not accept any touch input. 

Touch worked fine when I originally had Octoprint installed on this Pi and had a GUI installed on it to use the first iteration of Touch UI, but now I can't do anything with it.

I've quadruple checked the ribbon cable and lost count the amount of times I've reseated it, but it just doesn't want to accept any touch input.

snfsa0, 2022-10-13:
Hi, I have encountered the same issue with the original Raspberry Pi 7" touchscreen. The screen and touch funtionality works fines but after installing octodash it boots up properly with the "Hey There" welcome screen but screen isn't touch sensitive anymore.

oobedoob, 2022-10-13:
I found that my problem was the digitizer on the screen had gone bad. Flashed another SD card with Bullseye and the touch was only working on some parts of the very left edge of the screen (could only touch the "Start" button and some of the items on the menu). Got another screen and everything just worked.

closed 2025-03-18 by queengooborg