UnchartedBull/OctoDash #3234: Problem boot Octodash (RPI3B+Octoprint Lasted Ver.)

pauluchin, 2022-10-10:
**What doesn't work?**
The RPI starts normally but when starting Octodash there is a line left and it blinks a couple of times showing the interface and returns to the white line

**What did you already try?**
1- Clean IMG Octoprint 
2- Update & Upgrade
3- Install LCD-Show Drivers (3.5" XPT2046 model)
3B- Rotate screen 180°
4A- Directly Install Octodash  (Problem X not start)
4B- Add this ( [https://github.com/UnchartedBull/OctoDash/issues/1451#issuecomment-782796780](url) from issue #1451 and reboot. Octodash boot but with the problems described.
5- All steps in Troubleshooting guide :(

***
- When I try to calibrate the screen it gives an error of Unable to connect to X server
- VNC show the same img 

Link to video [https://youtu.be/5goJugLGrGM](url)

**General Information:**
 - Rpi 3B + Touch screen 3.5" XPT2046 model (GPIO)
 - OS Info The one they use in the latest Octoprint build
 - OctoDash Version 3.2.1
 - OctoPrint Version 1.8.3
[image]

**Additional context**
Add any other context or screenshots about the feature request here.

pauluchin, 2022-10-11:
Solved

closed 2022-10-11 by pauluchin

NGatti1997, 2022-11-30:
No information on what fixed it?