UnchartedBull/OctoDash #3268: Is it possible to close out of Octodash and return back to your RasberryPi Terminal?

OpazAlts, 2022-10-28:
Totally starting out here with all of this.  Fledgling at best when it comes to the Raspberry Pi and Octoprint/Octodash.  I was just wondering if it is still possible to quit Octodash and be able to return to the RPi terminal.

So far I feel like uninstalling Octodash is the only way to do it.  I read one article related to this but it was affecting the person’s screen.  Not really related to my question.   

**General Information:**
 - Raspberry Pi 4
 - Latest Raspian with latest Octoprint and Octodash loaded.  
 - OctoDash Version [e.g. v1.0.0] - Latest version
 - OctoPrint Version [e.g. v1.0.0] - Latest version 

I’d like to still be able to use the RPi terminal.  Any advice?

jneilliii, 2022-10-28:
It is possible to stop octodash a couple of different ways. The simplest is with my octodash companion plugin. With that installed it will add a stop octodash option to the system menu in OctoPrint.

OpazAlts, 2022-10-28:
But does it return you to your RasberryPi terminal?

jneilliii, 2022-10-28:
Yes

OpazAlts, 2022-10-28:
So after I select Stop OctoDash from the menu, OctoDash successfully quits but I get sent to the terminal window and cannot type anything.  I get a string of Error lines which I assume are related to shutting off OctoDash.  I'm unable to type anything.  All I have is the blinking curser at the bottom of the screen.  How do I get back to the Raspberry Pi terminal?  Thanks ahead of time.

jneilliii, 2022-10-28:
ctrl+t and ctrl+k I think will kill ratpoison and take you back to terminal

OpazAlts, 2022-10-28:
Thank you for the suggestion.  I tried it both through my Mac terminal (ssh) and with the RPi while Octodash is loaded.  Unfortunately, nothing happened.

closed 2025-03-18 by queengooborg