UnchartedBull/OctoDash #3285: I managed to kill my perfect running octodash with original raspberry 7'' display

ffutlapuztar, 2022-11-08:
**What doesn't work?**
I have a raspberry 3b running with an octopi image with a offical 7'' display for octodash- everything wasrunning without a problem for many month.
I then tried to convince the 7'' display to switch of backlight. I perhaps did some stupid things in raspi-config, but now switched everything back.

**What did you already try?**
The offical 7'' display was not connected by hdmi and I think there are no special divers needed.
Anyhow I managed to break the output on the lcd screen.

**General Information:**
 - Hardware Raspberry PI 3B
 - OS Info: OctoPi 0.17.0
 - OctoDash newest
 - OctoPrint 1.81
 - 
**Additional context**
I add here the content of the Xorg.o.log:

<sub>[    20.787] 
X.Org X Server 1.20.4
X Protocol Version 11, Revision 0
[    20.792] Build Operating System: Linux 5.4.0-54-generic armv8l Raspbian
[    20.794] Current Operating System: Linux octopi 5.10.11-v7+ #1399 SMP Thu Jan 28 12:06:05 GMT 2021 armv7l
[    20.794] Kernel command line: coherent_pool=1M 8250.nr_uarts=0 snd_bcm2835.enable_compat_alsa=0 snd_bcm2835.enable_hdmi=1 bcm2708_fb.fbwidth=800 bcm2708_fb.fbheight=480 bcm2708_fb.fbswap=1 vc_mem.mem_base=0x3ec00000 vc_mem.mem_size=0x40000000  console=ttyS0,115200 console=tty1 root=PARTUUID=6c586e13-02 rootfstype=ext4 elevator=deadline fsck.repair=yes rootwait
[    20.804] Build Date: 15 January 2021  02:03:57PM
[    20.806] xorg-server 2:1.20.4-1+rpt2+deb10u3 (https://www.debian.org/support) 
[    20.808] Current version of pixman: 0.36.0
[    20.812] 	Before reporting problems, check http://wiki.x.org
	to make sure that you have the latest version.
[    20.812] Markers: (--) probed, (**) from config file, (==) default setting,
	(++) from command line, (!!) notice, (II) informational,
	(WW) warning, (EE) error, (NI) not implemented, (??) unknown.
[    20.820] (==) Log file: "/var/log/Xorg.0.log", Time: Tue Nov  8 00:57:31 2022
[    20.825] (==) Using system config directory "/usr/share/X11/xorg.conf.d"
[    20.832] (==) No Layout section.  Using the first Screen section.
[    20.832] (==) No screen section available. Using defaults.
[    20.832] (**) |-->Screen "Default Screen Section" (0)
[    20.832] (**) |   |-->Monitor "<default monitor>"
[    20.838] (==) No device specified for screen "Default Screen Section".
	Using the first device section listed.
[    20.838] (**) |   |-->Device "Allwinner A10/A13 FBDEV"
[    20.838] (==) No monitor specified for screen "Default Screen Section".
	Using a default monitor configuration.
[    20.838] (==) Automatically adding devices
[    20.838] (==) Automatically enabling devices
[    20.838] (==) Automatically adding GPU devices
[    20.845] (==) Max clients allowed: 256, resource mask: 0x1fffff
[    20.848] (WW) The directory "/usr/share/fonts/X11/cyrillic" does not exist.
[    20.848] 	Entry deleted from font path.
[    20.848] (WW) The directory "/usr/share/fonts/X11/100dpi/" does not exist.
[    20.848] 	Entry deleted from font path.
[    20.848] (WW) The directory "/usr/share/fonts/X11/75dpi/" does not exist.
[    20.848] 	Entry deleted from font path.
[    20.848] (WW) The directory "/usr/share/fonts/X11/Type1" does not exist.
[    20.848] 	Entry deleted from font path.
[    20.848] (WW) The directory "/usr/share/fonts/X11/100dpi" does not exist.
[    20.848] 	Entry deleted from font path.
[    20.849] (WW) The directory "/usr/share/fonts/X11/75dpi" does not exist.
[    20.849] 	Entry deleted from font path.
[    20.849] (==) FontPath set to:
	/usr/share/fonts/X11/misc,
	built-ins
[    20.849] (==) ModulePath set to "/usr/lib/xorg/modules"
[    20.849] (II) The server relies on udev to provide the list of input devices.
	If no devices become available, reconfigure udev or disable AutoAddDevices.
[    20.849] (II) Loader magic: 0x1fdf80
[    20.849] (II) Module ABI versions:
[    20.849] 	X.Org ANSI C Emulation: 0.4
[    20.849] 	X.Org Video Driver: 24.0
[    20.849] 	X.Org XInput driver : 24.1
[    20.849] 	X.Org Server Extension : 10.0
[    20.851] (--) using VT number 2

[    20.851] (II) systemd-logind: logind integration requires -keeptty and -keeptty was not provided, disabling logind integration
[    20.851] (II) no primary bus or device found
[    20.852] (II) LoadModule: "glx"
[    20.855] (II) Loading /usr/lib/xorg/modules/extensions/libglx.so
[    20.929] (II) Module glx: vendor="X.Org Foundation"
[    20.929] 	compiled for 1.20.4, module version = 1.0.0
[    20.929] 	ABI class: X.Org Server Extension, version 10.0
[    20.929] (II) LoadModule: "fbturbo"
[    20.932] (WW) Warning, couldn't open module fbturbo
[    20.932] (EE) Failed to load module "fbturbo" (module does not exist, 0)
[    20.932] (EE) No drivers available.
[    20.932] (EE) 
Fatal server error:
[    20.936] (EE) no screens found(EE) 
[    20.938] (EE) 
Please consult the The X.Org Foundation support 
	 at http://wiki.x.org
 for help. 
[    20.945] (EE) Please also check the log file at "/var/log/Xorg.0.log" for additional information.
[    20.948] (EE) 
[    20.950] (EE) Server terminated with error (1). Closing log file.
</sub>

ffutlapuztar, 2022-11-26:
thnx for support.
solved it on my own.

jneilliii, 2023-10-20:
> thnx for support. solved it on my own.

what did you do to fix it? someone on Discord is having the same problem.

Sekisback, 2024-02-06:
after one year without problems i got the same behaviour after power off. octoprint is working but octodash wont start, due to this failure, didnt found a solution till now. Help needed