UnchartedBull/OctoDash #3297: Can't get past initializing screen without rebooting multiple times

TheFeralEngineer, 2022-11-14:
**Describe the bug**
When starting up pi controller (MKS Pi), octodash has about a 30% chance of starting properly. I don't have this issue when just running octoprint. 

**To Reproduce**
Steps to reproduce the behavior:
1. Power on MKS Pi
2. Octodash loads
3. Stuck on Initializing screen

**Expected behavior**
I want the thing to work! 

**Screenshots**

**General Information:**
 - MKS Pi running MKS Armbian distro
 -  OS: https://drive.google.com/drive/folders/1tTuSvF9OL2qtPXElau8YOXn2sWbdxa9e
 - Latest version of Octoprint and Octodash

**Additional context**
It's really getting on my nerves...

TheFeralEngineer, 2022-11-27:
Is this project dead?

jneilliii, 2022-11-27:
it's not dead, there's just not a full time team working on it. try SSH to your device and deleting the config file with the command `rm ~/.config/octodash/config.json` and then `sudo service getty@tty1 restart`. When entering your OctoPrint details, use the manual option to enter http://127.0.0.1:5000/.