UnchartedBull/OctoDash #3349: BTT TFT35 V3 LCD Screen

XeroZar, 2022-12-11:
Can this be installed on the BTT TFT35 V3 LCD Screen?

jneilliii, 2022-12-11:
If it's pi compatible then it should be possible. I use it with a BTT PiTFT50.

XeroZar, 2022-12-11:
> If it's pi compatible then it should be possible. I use it with a BTT PiTFT50. 

So it will replace the factory firmware with octodash?

jneilliii, 2022-12-11:
my specific touch screen doesn't have firmware as far as I know, it's just a display for the pi that utilizes the dsi ribbon connector on the pi for connection. sounds like you have a screen that is made more for connecting to printer control boards and act as a host control software to the printer's main board.

thegarbz, 2022-12-13:
jneilliii is correct. The BTT PiTFT50 is a control display largely targetted at the Ender3. This interface is not an interface for the 3D printer, but rather an interface for OctoPrint which is separate from your 3d printer firmware. https://octoprint.org/