UnchartedBull/OctoDash #3402: TouchScreen Controls Inverted

acer8888, 2023-01-15:
**What doesn't work?**
My LCD touchscreen touch points are upside down.
I was able to get the display to be orientated correctly, however having difficulties getting the touch points correct.

**What did you already try?**
I believe I need to set "xinput set-prop ‘ADS7846 Touchscreen’ ‘Coordinate Transformation Matrix’ -1 0 1 0 -1 1 0 0 1"

I opened a terminal on the display through Ratpoison (CTRL + T, C) ran the command, it accepted it but nothing changed. I also put the command in my $HOME/.xinitrc before OctoDash is launched, no change.

I tested with a normal Raspbian image and it works properly, I just cannot get it to work with Ratpoison and Octodash. Any help is appreciated.

**General Information:**
 - Hardware
 Raspberry Pi 4B - Generic 5in touchscreen LCD
 - OS Info
 Raspbian GNU/Linux 10 (buster)
 - OctoDash Version
 v1.8.6
 - OctoPrint Version
 Octopi 0.18.0

bokodimate, 2023-01-16:
I had the same issue. My screen is in vertical position. 
Open /boot/config.txt using PuTTY with the following code:
sudo nano /boot/config.txt

To rotate the screen (1 for 90 degrees, 2 for 180 degrees, 3 for 270 degrees) write this line:
display_lcd_rotate=3  

The following line fixed my touch screen points. I believe this swaps the X and Y coordinates. You can give it a try. 
dtoverlay=pri-ft5406,touchscreen-swapped-x-y=1,touchscreen-inverted-y=1

You can find more info here:
https://forums.raspberrypi.com/viewtopic.php?t=253973

... and here:
https://forums.raspberrypi.com/viewtopic.php?f=63&t=253959&p=1549853#p1549853

acer8888, 2023-01-17:
Hi @bokodimate 
Thank you for the input. I found those same articles you linked, it appears with Raspberry pi 4 the /boot/config.txt changes no longer work.. Due to an update video driver you have to use: 
DISPLAY=:0 xrandr --output HDMI-1 --rotate inverted

Here is the site I found the updated changes.. 
https://linuxhint.com/rotate-screen-in-raspberry-pi/

I really think the xinput set-prop command is the way to make the touchpoints correction, im just unfamiliar on how to make it work with octodash.