UnchartedBull/OctoDash #3441: [!WEB] and [!WEBCAM] commands not working from the Control screen

senghauser, 2023-02-03:
I added a custom [!WEB] command, and it works correctly if i open it while on the Notifications screen, however no iframe is displayed if I open it while on the Control page instead.

If I tap the command, then open the Navigation screen, it shows "loading website," but just hangs there. I need to close the browser window and then reopen it from the navigation screen. 

I have tried disabling the icon for the Notification Center, as well as toggling the confirmation and return to home screen options. I am getting the same behavior with the [!WEBCAM] command.

If i recall correctly, I believe the webcam worked for me the first time I tried it, when I first installed OctoDash, but It has not worked since. 

[image]
I tap the command from the control screen and get no iframe.

[image]
i then open the Notification center, and the iframe is in here, but wont load

[image]
I close the iframe and then click the command from this screen

[image]
And it loads with no issue.

Best I can tell, the iframe opens inside of the Notification Center screen. Is there some setting I am missing that allows the iframe to open outside of the Notification Center?

zach-ovic, 2023-03-17:
Hi, same issue here. Hopefully this will get fixed.

emelbardis, 2023-03-25:
Mine has never worked. Always figured my syntax was wrong
`[!WEB]http://localhost:5000/plugin/octodashcompanion/webcam`