UnchartedBull/OctoDash #3447: OctoDash is displayed but Touchscreen does not work on any part of the screen.

dbfrompw, 2023-02-05:
**What doesn't work?**
OctoDash is displayed but Touchscreen does not work on any part of the screen. 

**What did you already try?**
A clear and concise description of what you tried to make OctoDash work.

Using two different 7" HDMI touchscreens, a Pi3b and Pi4 2gb. The touchscreen I'm trying to use came with a Folgertech FT-6. Could not get it to work so I bought a new one from Amazon "https://www.amazon.com/dp/B0BG4ZCHCG?psc=1&ref=ppx_yo2ov_dt_b_product_details". Also did not work. Followed the instructions 5 times with no joy. 
**General Information:**
 - Hardware [e.g. Raspberry Pi, if you have layout issues please also include your screen resolution]
 - OS Info [e.g. Raspbian Buster Lite, please also indicate if you used OctoPi]

I used Raspberry Pi Imager to make the Octoprint SD card.
[image]
[image]

-OctoDash version : fresh install as of 2/4/23
-OctoPrint version : 1.8.6
-OctoPi version    : 0.18.0

**Additional context**
Add any other context or screenshots about the feature request here.

I can send you the Touchscreen to try it out.

ThunderClap86, 2023-04-05:
I am having the same issue on a RPi 7" screen. Virtually same information the OP has above.

Drew3164, 2023-04-07:
I have the same problem, is there any fix on this???

dbfrompw, 2023-04-07:
Turns out that you can’t have external power and if your power supply goes below 5v at all, it doesn’t work.

KevinGroninga, 2023-04-07:
My RPi 7" TOuchscreen is wired (powered) by the small jumpers from the screen to the actual RPi board (5v).  So in this case, it's not externally powered and my Touchscreen barely works!  I have to tap everything about a dozen times before it recognizes that it's been touched.   

 Hoping that someone at least will provide some feedback or useful info on why this setup doesn't work.  I can see OctoDash being useful, but not if the touchscreen isn't working.  Then it's just mere;y a display that you really can't do much with.

Drew3164, 2023-04-07:
For me it has nothing to do with 5v power as I had an old SD card with Octoprint and Octodash that i just tested to see if it works and yes the touchscreen works fine. But going ahead and reinstalling a new octoprint and new octodash touchscreen is not working?

VisualReversal, 2023-04-30:
I’m in the same boat as everyone else here. I’m using the HMTECH 10.1” hdmi touchscreen. It works fine as a touchscreen for everything except octodash. Like everyone else, it’ll display the octodash screen but the touch functions don’t work.

Lampyridae1984, 2023-07-05:
Possible solutions: [#3559](https://github.com/UnchartedBull/OctoDash/issues/3559#issuecomment-1503525080) & [#3714](https://github.com/UnchartedBull/OctoDash/issues/3714#issuecomment-1620899774)

KevinGroninga, 2023-07-05:
Awesome!   I may have to try both of these solutions.  Sometimes the OctoDash screen is responsive, but only when the printer isn’t actively running a job.  So I’ll try both solutions and in between each, run a job and see if the screen is working as we would expect them to.

 

Thanks again!   I’ll definitely get back to you with the results of my testing!