UnchartedBull/OctoDash #3486: No response after command line entered

berb6, 2023-02-27:
Hello, 

I have octopi set up, latest stable version 1.0.0. Set this up on Friday 2/24/2023. 

I have been trying to set up my octodash on my RSp 3B using a 5" Elecrow touch screen GPIO TFT screen.
- link here [
](https://www.elecrow.com/hdmi-5-inch-800x480-tft-display-for-raspberry-pi-b-p-1384.html)

I have tried about 15-20 youtube variations to get octodash to actually appear. 

I have octodash installed on my .local via the web interface. 

For the love of me I can not get it installed on the pi vs SSH. 

Used: [
](https://unchartedbull.github.io/OctoDash/index.html)

Tried  [
](https://unchartedbull.github.io/OctoDash/index.html)

I get no response on the terminal it just goes stagnant and have to command C to exit terminal line and start over. 

[
](https://howchoo.com/octoprint/3d-printer-octoprint-touchscreen)

[image]

I'm sure I left things out but I am very new to this and ready to quit and return all the gear. Octopdash install is giving me so much trouble. I can't even link all the youtube videos and websites it prompted me to on how they installed. I finally landed on your page and seemed legit and the best but as said above the main installer code just freezes and no activity on the terminal.

berb6, 2023-02-27:
[[]](https://unchartedbull.github.io/OctoDash/index.html)(url)

berb6, 2023-02-27:
[image]

berb6, 2023-02-27:
I also tried the manual way (https://github.com/UnchartedBull/OctoDash/wiki/Installation#manual-installation) and get this error wget: unable to resolve host address ‘gir1.2-gnomekeyring-1.0~’

berb6, 2023-02-28:
I found this link that detials basically the same issue I am having. I checked my Xterm-color both in SSH and out out of. Both are <xterm-256color> (https://github.com/UnchartedBull/OctoDash/issues/1335)

berb6, 2023-02-28:
I belive I have octoprint installed? 

[image]

I did follow the above link and change master to main.

jneilliii, 2023-02-28:
I suspect this issue is related to the latest underlying OS the new OctoPi 1.0.0 has, which is Raspbian OS (Bullseye). The previous OctoPi image used the older Buster version of Raspberry Pi OS.

berb6, 2023-02-28:
@jneilliii Is there a way around this? It did take quite some time to set up 1.0.0 .. about 2 days of various methods before it worked.

jneilliii, 2023-02-28:
#2531 would be where to watch.

berb6, 2023-02-28:
@jneilliii thank you .. still at a loss even after following #2531

jneilliii, 2023-02-28:
The way I understand it, after the install you need to edit the file `.xinit.rc` and make it look like this, the important bit beeing 
`--no-sandbox`

`nano ~/.xinitrc`
```
#!/bin/sh

xset s off
xset s noblank
xset -dpms

ratpoison&
octodash --no-sandbox
```

berb6, 2023-02-28:
Where is this file `.xinitrc`?

I installed `sudo /home/pi/scripts/install-desktop`

jneilliii, 2023-02-28:
no, after you run the OctoDash install script. after editing the file with that command you then press ctrl+x to exit, answer y to save and enter to keep same filename and then `sudo reboot`.

berb6, 2023-02-28:
Heres the issue, the install scrip does not work for me `wget (https://github.com/UnchartedBull/OctoDash/raw/main/scripts/install.sh)`

berb6, 2023-02-28:
I guess I'm not sure how to fix this so I can run OctoDash. 

Sould I copy the installer code and change to --no-sandbox then run it again?

jneilliii, 2023-02-28:
I was planning on upgrading one of my instances this weekend to do some testing. Just off the top of my head I would try to do the manual install but remove that one package from the apt install command.
```
sudo apt install libgtk-3-0 libnotify4 libnss3 libxss1 libxtst6 xdg-utils libatspi2.0-0 libuuid1 libappindicator3-1 libsecret-1-0
```
then proceed with the remaining steps of the manual install, but include that bit when creating the .xinitrc file.

berb6, 2023-02-28:
[image]

berb6, 2023-02-28:
Tried the instance you mentioned and another error
[image]

Hillshum, 2025-03-21:
Please try with the latest release of OctoDash