UnchartedBull/OctoDash #3559: Not working with OctoOi 1.0.0

ordor2k, 2023-04-02:
**Describe the bug**
A clear and concise description of what the bug is.
New install of OctoPi 1.0.0 with OctoPrint 1.8.7 after apt update and upgrade, the installation finish successfuly but the app is not working.
**To Reproduce**
Steps to reproduce the behavior:
1. get a new OctoPi installation with the Pi imager tool
2. do "sudo apt update && sudo apt upgrade -y"
3. try install with the script "bash <(wget -qO- https://github.com/UnchartedBull/OctoDash/raw/main/scripts/install.sh)"
4. the installation finished successfully and after reboot there are few errors regarding x.org x server and more.

**Expected behavior**
after the installation and a reboot the octodash screen should work

**Screenshots**
https://photos.app.goo.gl/2CVK3LmChLJz4Zo58

**General Information:**
 - Hardware Raspberry Pi 4B 2G, BIGTREETECH PITFT50 v1
 - OS Info  OctoPi 1.0.0
 - OctoDash Version 2.3.1
 - OctoPrint Version 1.8.7

**Additional context**
Add any other context about the problem here.

taclane, 2023-04-06:
I just found myself stuck in this too. I haven't figured why the installer is having an issue with the arm64 package, but after running the script, ssh back in to the Octopi install and manually install the armv7l version.

```
wget https://github.com/UnchartedBull/OctoDash/releases/download/v2.3.1/octodash_2.3.1_armv7l.deb
sudo dpkg -i octodash_2.3.1_armv7l.deb
```

Once that finishes, reboot the pi, and and the Octodash setup should continue. Ideally, Octopi 1.0 should be using the arm64 version, but until that's sorted out the 32-bit version seems to work fine.

jneilliii, 2023-04-07:
> Ideally, Octopi 1.0 should be using the arm64 version

Fairly confident that the OctoPi image from pi imager is still 32 bit. Did you install the 64 bit version?

taclane, 2023-04-07:
It's a little slow going, but I'm double-checking the arm64-only version right now.  Took a moment to even make sense of what was going on with the default version offered by the pi imager, because after some updates, it looked like it was using the 64bit.

`Linux octopi 6.1.21-v8+ #1642 SMP PREEMPT Mon Apr  3 17:24:16 BST 2023 aarch64`

taclane, 2023-04-07:
> Fairly confident that the OctoPi image from pi imager is still 32 bit. Did you install the 64 bit version?

It seems like that was it.  After just *apt updating* the OctoPi 1.0.0 image provided by the official tool, I think my Pi4 ended up with a mismatched set of arm64 and armhf files that prevented OctoDash from installing correctly.  

No such issues after starting with the base arm64 image direct from the OctoPi repo.
https://github.com/guysoft/OctoPi/releases

I can't be entirely certain that my issue was the same as the others above, but if they don't have a `/usr/bin/octodash`, it's probably worth switching over to the 64-bit image and trying again.

twinkie0101, 2023-04-07:
I'm out of town for the weekend but will try this when I get home to see if it solves the issue for me as well, thanks for digging in to this issue.

twinkie0101, 2023-04-10:
Just tried installing the base arm64 image from the repo and installed octodash with zero issues. Working as expected.

jneilliii, 2023-04-10:
> It could not install gir1.2-gnomekeyring

Yeah, I don't think that's required anymore and the manual install instructions haven't been updated.

ordor2k, 2023-04-10:
> Just tried installing the base arm64 image from the repo and installed octodash with zero issues. Working as expected.

I did the same and now its working, but the touch screen isn't.
Octodash on the old octopi was working with no issue.

twinkie0101, 2023-04-10:
@ordor2k do you have the official raspberry pi screen? If not, I believe you'll need to install the appropriate driver for the display.

taclane, 2023-04-11:
> Fairly confident that the OctoPi image from pi imager is still 32 bit. Did you install the 64 bit version?

To put some closure to this: 
> The updated firmware now prefers the 64-bit kernel on a Pi 4 because it should give better performance in most situations. To continue to boot to the 32-bit kernel, add `arm_64bit=0` to config.txt.

https://github.com/raspberrypi/linux/issues/5402#issuecomment-1484676216

The Octopi 32-bit image used to work fine as of a little while ago, but after a recent change in raspbian, it's now pushing a 64-bit kernel on top of the 32-bit userspace, and that's breaking the octodash installer's ability to correctly figure out which .deb to use.
https://github.com/UnchartedBull/OctoDash/blob/d2134ea873410146133e667596e705888c0a6126/scripts/install.sh#L671

taclane, 2023-04-11:
> Working good now, THANKS!!!
Great!
An alternate option is to keep the new drivers but force it to load the overlay for 7-inch displays if it isn't detecting correctly:
```
# Enable DRM VC4 V3D driver
dtoverlay=vc4-kms-v3d
dtoverlay=vc4-kms-dsi-7inch
```
Though, with the new drivers the touchscreen doesn't feel as responsive with whatever hardware revision of raspi display I have, so I've personally had a better experience disabling kms altogether.

Extra-Fox, 2024-01-06:
> * ssh in to the Octopi install
> * `sudo nano /boot/config.txt`
> * find
> 
> ```
>  # Enable DRM VC4 V3D driver
>  dtoverlay=vc4-kms-v3d
> ```
> 
> * comment out the second line
> 
> ```
> # Enable DRM VC4 V3D driver
> #dtoverlay=vc4-kms-v3d
> ```
> 
> * reboot

This was exactly what worked for me to get my Waveshare 5 inch DSI TFT display to work. 

I just hope I remember this whenever I have to go reset this up.

closed 2025-03-21 by Hillshum