UnchartedBull/OctoDash #3564: No fan speed value

CLNilix, 2023-04-06:
**What doesn't work?**
on the main print screen of OctoDash, the fan speed value is blank, although the dashboard of Octoprint shows the correct value.

**What did you already try?**
nothing.

**General Information:**
 - Hardware: Raspberry Pi Zero 2 W Rev 1.0
 - OS Info: OctoPi Version 1.0.0
 - OctoDash Version 2.3.1 ? (automated script installed a few days ago)
 - OctoPrint Version 1.8.7

**Additional context**
there is also a value of -1 for the fan speed in the "adjust" screen.

[image]
[image]
[image]

Mikec78660, 2024-06-16:
I just installed Octodash a few days ago and came here to look for help with the exact issue. I can control the fan fine with gcode in the terminal so I know I have the correct pin and everything. But other than this fan speed issue I love having a touch screen on the printer since installing klipper. Even if i could get the lcd screen back working on the printer (ender3v2) I probably wouldn't, Octodash is so much better. Thank you so much for working this!

queengooborg, 2025-03-18:
The fan speed is obtained through the DisplayLayerProgress plugin, or the OctoDash Companion plugin since OctoDash v2.4.0 (release pending at time of writing).  You will need to have either of these plugins installed and support enabled in OctoDash's settings to view and configure fan speed!

closed 2025-03-18 by queengooborg