UnchartedBull/OctoDash #3572: Touchscreen not responding ?

Drew3164, 2023-04-08:
I have reinstalled OctoPrint and reinstalled OctoDash not the touchscreen is not responding ??

Is there a way to calibrate the touchscreen ?

Thanks

taclane, 2023-04-11:
Check out some of the discussion over here:
https://github.com/UnchartedBull/OctoDash/issues/3559#issuecomment-1503525080

It's possible that you're just having an issue with the display driver after getting a new system online. Editing a line or two in config.txt might sort things out.

Teslafly, 2023-05-26:
I had an issue with i2c not being enabled for the touchscreen touch controller.
https://github.com/OctoPrint/OctoPi-UpToDate/issues/3#issuecomment-1563672255

Lampyridae1984, 2023-07-05:
[#3559](https://github.com/UnchartedBull/OctoDash/issues/3559#issuecomment-1503525080) & [#3714](https://github.com/UnchartedBull/OctoDash/issues/3714#issuecomment-1620899774)

queengooborg, 2025-03-18:
https://github.com/UnchartedBull/OctoDash/wiki/Troubleshooting#touchscreen-is-not-responding

closed 2025-03-18 by queengooborg