UnchartedBull/OctoDash #3579: Touchscreen stops working when enabling the camera

seenev, 2023-04-12:
**What doesn't work?**
After getting OctoDash to work and display on the touch screen, I connected and enabled the camera. I can see that the camera works from OctoPrint on my computer, but now the touchscreen is blank.

**What did you already try?**
I tried modifying the gpu_mem=128 value in the config file. That will make the touchscreen work, but then the camera does not work.

**General Information:**
 - Raspberry Pi 4 [e.g. Raspberry Pi, if you have layout issues please also include your screen resolution]
 - OctoPi 1.0.0
 - OctoDash Version 2.0.0
 - OctoPrint Version [e.g. v1.8.7]

seenev, 2023-04-14:
I was able to get this working by modifying values in the config file. I don't recall exactly which value it was. I have found that I can only get the screen to display with hdmi_group=1 and hdmi_mode=1, but the OctoDash screen looks kind of skewed when shifted for landscape and it is missing info like the time left or the preview icon. I did upgrade to 2.3.1 but it still doesn't look right. Is there something you need to do to get it to look right in landscape mode?

Ruhel786, 2023-04-14:
What screen are you using? Will help to see which settings you need.
Also same with your camera.
Thanks

seenev, 2023-04-14:
Display: 3.5 Inch RPi Display Touch - XPT2046 (480x320)
I'm just using the standard Raspberry pi camera I've had for years and years. Probably a rev1. The camera is working now. The only issue I have at this point is that it looks like the screen is not at the right aspect ration. The settings gear at the top corner of OctoDash is half off the screen at the top.

Ruhel786, 2023-04-14:
When you first got OctoDash working before attaching the camera, was the aspect ratio correct?
It should be able to adjust its self according to this thread.
https://github.com/UnchartedBull/OctoDash/issues/145

seenev, 2023-04-14:
Before attaching the camera, it looked OK but it was in the portrait mode. I rotated the display to mount it to the printer and that's when it started looking wonky.

Ruhel786, 2023-04-14:
This is my screen at the moment but I'm using the HyperPixel Screen. Can you try going back to the original config.txt. 
[image]