UnchartedBull/OctoDash #3600: octodash package architecture (arm64) does not match system (armhf)

HMRudi, 2023-04-23:
**What doesn't work?**
Installation of octodash goes wrong. see screenshot

**What did you already try?**

Ignored error first, but octodash is not starting at all. I was searching for the error, but all abswers are confusing me. Do I have the wrong installer package, wrong OS or wrong Hardware?

**General Information:**
 - Raspberry 4
 - OS Info: OctoPi OS
 - OctoDash Version 2.3.1
 - OctoPrint Version 1.8

**Additional context**
[image]

Add any other context or screenshots about the feature request here.

VisualReversal, 2023-04-24:
Thank god! I thought I was the only one dealing with this. I'm getting exactly the same on my RPI 400. I've tried everything and am completely stuck. Not sure if (or how) this is related, but this is what comes up on my hyperpixel display. When I boot up. 
[image]

HMRudi, 2023-04-24:
@VisualReversal 
Thank you for your reply. I am using the 7" Panel [https://www.amazon.de/Raspberry-Pi-7-Inch-Screen-Display/dp/B014WKCFR4/ref=sr_1_10?keywords=7+Zoll+Raspi+panel&link_code=qs&qid=1682353547&sourceid=Mozilla-search&sr=8-10 ](url) Is our problem panel related?
I'am not as far as you did. Right now I am struggeling with basic things. Cannot install **gir1.2-gnomekeyring-1.0**
I don't know why.  Manual Installation says:
```
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
E: Unable to locate package gir1.2-gnomekeyring-1.0
E: Couldn't find any package by glob 'gir1.2-gnomekeyring-1.0'
```
I tried to install "the Girl" manually
```
sudo apt-get update
sudo apt-get install gir1.2-gnomekeyring-1.0
```
Answer:
```
Reading state information... Done
E: Unable to locate package gir1.2-gnomekeyring-1.0
E: Couldn't find any package by glob 'gir1.2-gnomekeyring-1.0'
E: Couldn't find any package by regex 'gir1.2-gnomekeyring-1.0'
```
I am Confused??!!
Best regards

VisualReversal, 2023-04-24:
I wasn’t able to install that particular dependency either, so I skipped it. Apparently it wasn’t necessary in my case.

VisualReversal, 2023-04-28:
Not sure why I didn't figure this out sooner, but I was able to apply the latest update by simply substituting the url for version 2.0.0 with the url for version 2.3.1 (or alternatively replacing "2.0.0" with "2.3.1" in the two spots in the url listed in the manual instructions - see highlighted part of attached screen shot). Hopefully this helps someone down the road. 

[image]