UnchartedBull/OctoDash #3601: 409 conflict error appears on Octodash screen.

dpm0, 2023-04-24:
I can't get Octodash to work. I continually get "can't send GCode" errors when trying to access the touchscreen that I have setup on my RPI4. In capital letters at the bottom of the error it says "409 CONFLICT" 

I am able to access the printer and print jobs from a browser using Octoprint on a Pc which is on the same internal network. The only connection to the printer is via the USB cable, which is between the printer and the PI which has the image of Octoprint on it (as well as Octodash). The Octoprint image is from the official RPI image installer.
Octoprint says the printer is online.

**General Information:**
 - Hardware Raspberry Pi4 /4mb ram 4" chinese touch screen from Ali.
 - OS Info: Raspbian Buster and OctoPi]
 - OctoDash Version is the one that "bash <(wget -qO- https://github.com/UnchartedBull/OctoDash/raw/main/scripts/install.sh)" installs.
 - OctoPrint Version Installed: 1.8.7

**Additional context**
[image]

dpm0, 2023-04-24:
my email address is dm@carolynmaher.com.au

Yoanf26, 2024-02-02:
I had the same problem. And it turns out that I made a mistake in the order. Once this error is corrected, the problem is no longer present.

closed 2025-03-18 by queengooborg