UnchartedBull/OctoDash #3671: OctoDash Touchscreen Not Working Octopi 1.0.0

whoha007, 2023-05-29:
Newest (stable) version of Octopi 1.0.0, and newest version of OctoDash stuck on Hey there! settings screen, cannot click next, no touchscreen response. Running Pi 4 on 7in official Pi touchscreen. I have older versions of Octopi running OctoDash on two other printers with the same setup and they work fine. The problem seems to be with the newest release of Octopi. The touchscreen I am using has been tested with another OS and works fine. Any help is appreciated!

alberto67leo, 2023-05-30:
Same problem for me, now I discovered that it’s not my fault, I was going crazy 😟. Any workaround? Maybe downgrade Octopi?

whoha007, 2023-05-30:
> Same problem for me, now I discovered that it’s not my fault, I was going crazy 😟. Any workaround? Maybe downgrade Octopi?

That's exactly what I did. I went down to 0.18.0, and it worked fine. The moral of the story is that they need to upgrade OctoDash to work with the latest version. It hasn't been updated for over a year. Or at the very least state the incompatibility issue in their FAQ.

whoha007, 2023-05-30:
> Same problem for me, now I discovered that it’s not my fault, I was going crazy 😟. Any workaround? Maybe downgrade Octopi?

That's exactly what I did. I went down to 0.18.0, and it worked fine. The moral of the story is that they need to upgrade OctoDash to work with the latest version. It hasn't been updated for over a year. Or at the very least state the incompatibility issue in there FAQ.

alberto67leo, 2023-05-30:
Thanks a lot, I will follow you doing the same downgrade. Have a nice day!

TheLivingBubba, 2023-06-05:
Been having the same problem with a Pi4b and a 7" original touch screen. All of the problems started when I updated pi os to Bullseye or whatever the latest is. 

If you were both successful in downgrading and everything working again, would you kindly point me to the files that worked?

whoha007, 2023-06-05:
https://unofficialpi.org/Distros/OctoPi/old/octopi-buster-armhf-lite-0.18.0.zip

This one should work. But I haven't tested it. I still had an old version on my local drive.

whoha007, 2023-06-05:
This is from a reply I got on Discord, but haven't had a chance to try it. Let me know if you guys have success. 

I had same issue. This fixed it -->https://github.com/UnchartedBull/OctoDash/issues/3559#issuecomment-1503525080

TheLivingBubba, 2023-06-05:
Thank you very much, I have tried commenting out the kms driver, adding the fkms driver, adding the driver for the DSI-7inch screen, added fpt5406, commented out things. I've edited the bootloader, etc. I can't get it to rotate the LCD and accept touch inputs. (All of which were fine before I stupidly updated.) 

Want to start fresh with a known good version and see what I messed up along the way.

Lampyridae1984, 2023-07-05:
Check out my solution in issue [#3714](https://github.com/UnchartedBull/OctoDash/issues/3714), perhaps it works.

rongith, 2023-10-15:
Octodash is working on a fresh new install for me.
Hardware:
- 	RPI4
- 	RPI official 7" touchscreeen

What I've done:
# Octopi
	copy octopi 1.0.0 image

# Sudo
	sudo nano /etc/sudoers.d/010_pi-nopasswd
		remove # (uncomment)
		
# Update
	sudo apt update
	sudo apt upgrade
	sudo reboot
	
# 7inch RaspberryPiScreen
	sudo nano /boot/cmdline.txt
		Add at the end of the line: video=DSI-1:800x480@60,rotate=90
	sudo nano /usr/share/X11/xorg.conf.d/40-libinput.conf
		For 180° add: Option "TransformationMatrix" "-1 0 1 0 -1 1 0 0 1"
		Section "InputClass"
			Identifier "libinput touchscreen catchall"
			MatchIsTouchscreen "on"
			Option "TransformationMatrix" "-1 0 1 0 -1 1 0 0 1"
			MatchDevicePath "/dev/input/event*"
			Driver "libinput"
		EndSection

# Configuration Octoprint

# Octodash
	Install
		bash <(wget -qO- https://github.com/UnchartedBull/OctoDash/raw/main/scripts/install.sh)

wangyao127, 2024-06-21:
> 我也遇到了同样的问题，现在我发现这不是我的错，我快疯了😟。有什么解决办法吗？也许可以降级 Octopi？

me too .But i try 0.17.0.IT NOT RUNNING ON RASPI 4b.

closed 2025-03-18 by queengooborg