UnchartedBull/OctoDash #3707: BTT pi TFT50 Touch Screen not working...

garylefebvreg, 2023-06-15:
**What doesn't work?**
Loaded OctoDash and it works but I have no touch screen capabilities.

**What did you already try?**
Everything I can think of. Google, Google, Google...

**General Information:**
 Hardware [e.g. Raspberry Pi4B
 
 OctoPrint Version [e.g. v1.0.0]

**Additional context**
The problem could be OctoPrint download octopi-1.0.0-1.9.0-20230523082504.img

TheLivingBubba, 2023-06-17:
Can you SSH into the Pi? If so you are going to need to edit the /boot/config.txt file to load the touch drivers. 

I would also go into raspi-config and enable vnc and you can finish octodash setup from a computer till you figure out your drivers. 

https://github.com/UnchartedBull/OctoDash/issues/3559#issuecomment-1503525080

Minora72, 2023-06-20:
I have two extruders and two Pi (3b_ and 4)...
After several hours and reimaging and reading online, found the problem with mine to be in the Config.txt. 

I changed:
dtoverlay=vc4-kms-v3d
to
dtoverlay=vc4-fkms-v3d  - which worked in the Pi Desktop... once...

After several more attempts, found it was missing:

 dtoverlay=rpi-ft5406
  
That got it working on mine - this morning (6/19/2023) (Pi3b+).  I am now setting up my second Pi (4) and will see if I get the same error.

FatCatMedia, 2023-07-09:
Hey so I had the same issue with my tft50 v2.0 and the dtoverlay=vc4-fkms-v3d  worked for me to get touch function working in octodash but now my issue is that the touch is way off. I try to touch "filament" and the "control" will open. It seems my touch is off and not lining up with the icons on display. How can I fix that?    (raspberry pi4b, newest versions of octopi and octodash, will be contolling a BBT octopus board connected to an ender 3 pro(overkill I know it was on sale)

FatCatMedia, 2023-07-09:
Okay figured it out on my own lol. Delete  dtoverlay=vc4-fkms-v3d   and replace with dtoverlay=rpi-ft5406 and all fixed and working. 3 days later of googleing lol.  So for anyone else doing this, setup octpi on a raspberry pi, follow octodash install instructions and install octodash, then in a ssh window type in   sudo nano /boot/config.txt   and it will open a config. Use you keyboard arrow keys to find dtoverlay=vc4-kms-v3d  delete the line and replace it with dtoverlay=rpi-ft5406  press control+x to exit, press y to save press enter to continue and then sudo reboot to restart pi and you will be in business.

garylefebvreg, 2023-07-10:
Yes sir, I used bits of this set of instructions

https://docs.google.com/document/d/1VfDyTq252wkcG9_D4Nh2lTRCSq51iltMDkG5AJ3J9jQ/edit

danielgardner2004, 2023-12-10:
hi, i dont have these lins in the config file do i just add them?

Vance1122, 2024-05-18:
I was having this issue with a official 7" raspberry pi touch screen. Using the dtoverlay=rpi-ft5406 instead of dtoverlay=vc4-fkms-v3d fixed it for me also. I didn't delete dtoverlay=vc4-fkms-v3d, I just commented it out like this -  #dtoverlay=vc4-fkms-v3d.

tdarlic, 2024-11-08:
Just to put this here. I had a same problem with my BTT 50. It was working perfectly and then probably after Klipper update touch stopped working. I tried reinstalling and whatnot with klipper bc I thought Klipper or some driver was the issue. I finally tried changing dtoverlay to rpi-ft5406 and it works better that before. 
Just to warn the noobs here as I've seen this being asked on forums often you need to change this in your /bootfs not in your printer configuration. The file name is config.txt