UnchartedBull/OctoDash #3714: Raspberry Pi 7' screen not responding to touch.

LynxFanta, 2023-06-18:
**What doesn't work?**
A clear and concise description of what the problem is. Ex. I'm always frustrated when [...]
7' Inch Raspberry Pi Touchscreen, touch display not responsive.
**What did you already try?**
A clear and concise description of what you tried to make OctoDash work.
Updating OctoDash & Print with running through previous builds.
Similar forums started recently within the past 2 weeks of other users having similar issues on their 7' Display

**General Information:**
 - Hardware [e.g. Raspberry Pi, if you have layout issues please also include your screen resolution]
  Raspberry Pi 4b
 - OS Info [e.g. Raspbian Buster Lite, please also indicate if you used OctoPi]
 Raspberry OS, OctoPi v1.0.0
 - OctoDash Version [e.g. v1.0.0]
 v1.9.0
 - OctoPrint Version [e.g. v1.0.0]
v2.3.1
**Additional context**
Add any other context or screenshots about the feature request here.
Other users experiencing similar issue
https://community.octoprint.org/t/touchscreen-not-responding-to-touch/52225/6

tcbarnett1996, 2023-06-25:
I am having the same issue. I have been reading forums for the last 3 hours. It seems to be a problem with an update that occured in September 2022. There are lots of posts asking for solutions, but no solutions. Ironically, I loaded OctoKlipper and it did not have this issue. I think I will switch to OctoScreen. Hopefully I will not encounter this issue with OctoScreen.

Lampyridae1984, 2023-07-05:
OMG, after searching for hours (!!!) online, I found a solution on a Klipper forum.

This worked for me:

(1) SSH into the Raspberry Pi
(2) Command `sudo raspi-config`
(3) Enabled SPI and I2C
(4) Saved and rebooted
(5) SSH again into the Raspberry Pi
(6) Command `sudo nano /boot/config.txt`
(7) Changed the line `dtoverlay=vc4-kms-v3d` to `dtoverlay=vc4-fkms-v3d`
(8) Saved changes and rebooted

Hope this works for anyone else. 🤞

KevinGroninga, 2023-07-05:
Wow! After banging my head on this for the last few months and sort of giving up on ever using OctoDash while a print job was running, this solution of going back to the correct driver finally solved the issue I was having!  Now I can finally enjoy being able to use OctoDash and the touchscreen to tune jobs while they are running instead of driving through the Marlin menu on my CR10 v3.  Oh happy days!

queengooborg, 2025-03-18:
This has been added to the wiki as a troubleshooting step!

closed 2025-03-18 by queengooborg